micropython/micropython #6950: esp32:  CMake not recognizing menuconfig output

bertelschmitt, 2021-02-23:
I am compiling for GENERIC_SPIRAM. The new CMake version requires (among other things) the size of the flash memory to be set via menuconfig. However, on running  make, the settings written to …/build/sdkconfig  are not reflected in  .../build-GENERIC_SPIRAM/sdkconfig  or   .../build-GENERIC_SPIRAM/sdkconfig.combined.
For instance, setting the flash size to 16MB in menuconfig produces this  in …/build/sdkconfig  

```
# CONFIG_ESPTOOLPY_FLASHSIZE_1MB is not set
# CONFIG_ESPTOOLPY_FLASHSIZE_2MB is not set
# CONFIG_ESPTOOLPY_FLASHSIZE_4MB is not set
# CONFIG_ESPTOOLPY_FLASHSIZE_8MB is not set
CONFIG_ESPTOOLPY_FLASHSIZE_16MB=y
CONFIG_ESPTOOLPY_FLASHSIZE="16MB"
```

However, the pertinent entries in …/build-GENERIC_SPIRAM/sdkconfig read like this:

```
# CONFIG_ESPTOOLPY_FLASHSIZE_1MB is not set
# CONFIG_ESPTOOLPY_FLASHSIZE_2MB is not set
CONFIG_ESPTOOLPY_FLASHSIZE_4MB=y
# CONFIG_ESPTOOLPY_FLASHSIZE_8MB is not set
# CONFIG_ESPTOOLPY_FLASHSIZE_16MB is not set
CONFIG_ESPTOOLPY_FLASHSIZE="4MB"
CONFIG_ESPTOOLPY_FLASHSIZE_DETECT=y
```

This causes compilation of code for 4MB, which is not desired.
Apparently, make picks up defaults from /esp32/boards/sdkconfig.base  while ignoring the settings produced by menuconfig.  

If I make the following edit in /esp32/boards/sdkconfig.base, the compile finishes successfully:

```
#CONFIG_ESPTOOLPY_FLASHSIZE_4MB=y
CONFIG_ESPTOOLPY_FLASHSIZE_16MB=y
```

However, this is not the desired solution

bertelschmitt, 2021-02-25:
My current workaround is to replace .../esp32/boards.skdconfig.base with the output of menuconfig, which seems to work, but I'm not sure how well.

wangshujun-tj, 2021-04-13:
Modify config. Base and partition table can be compiled successfully
However, after burning, the startup will report the following error:
entry 0x4008063c
[0;31mE (57) flash_parts: partition 3 invalid - offset 0x200000 size 0xe00000 exceeds flash chip size 0x400000[0m
[0;31mE (57) boot: Failed to verify partition table[0m

dpgeorge, 2021-04-13:
If you have a custom board then the idea is that you would create a custom board definition (eg copy boards/GENERIC to boards/MYBOARD and change as needed).  That way you can use the output of menuconfig to create the custom sdkconfig for your board.