micropython/micropython #6952: Question: Debugging micropython c/c++ code using VSCode

ALEX010001, 2021-02-23:
Hi,

I want to contribute to Micropython and Im looking for visual debugging tool, such as debugging with gdb on vscode that integrate with micropython.
 how can I do it?

stinos, 2021-02-23:
Not possible. See e.g. https://github.com/micropython/micropython/issues/6799.
Note this is a bug tracker so not really suitable for general questions, please search the forum for that: forum.micropython.org/

miketeachman, 2021-02-23:
> Im looking for visual debugging tool

The Forum contains many posts on debugging with gdb.  If you don't find what you need there, post a question on the Forum.

Please close this Issue to help maintain a relevant issues list.

mocleiri, 2021-02-28:
Its possible to debug the c/c++ side if you build the unix port.  I don't have it documented at all but this is my working launcher: https://github.com/mocleiri/tensorflow-micropython-examples/blob/master/.vscode/launch.json

The stopAtEntry flag needs to be true otherwise it doesn't stop at any breakpoint. I have two launchers they both refer to the micropython binary but specify different scripts to be run.

But this is only of limited use.  I've only used it with the unix port.  I don't know how to do it using a boards jtag interface.  This only works for the c/c++ side so you are still blind to most state and the calltrace of what is being executed on the script side.

stinos, 2021-02-28:
> The stopAtEntry flag needs to be true otherwise it doesn't stop at any breakpoint

That's odd, don't recall having a problem with that. So you're saying that for every C or C++ program you debug with VSCode, you need stopAtEntry otherwise no breakpoints are hit at all? Sounds like a bug. 

> But this is only of limited use

Depends on what you're doing. Has been extremely valueable for me. IIRC on the forum it has been confirmed that debugging microcontrollers over JTAG works as well.

> you are still blind to most state and the calltrace of what is being executed on the script side

Depends on how well you know the internals. Execution is in the end just one big loop executing bytecode, see vm.c, so if you understand some of what that does you can see what is going on. So yes it's not like stepping through Python code with watch windows etc, but depending on what you're debugging it can be useful.

ALEX010001, 2021-03-03:
finally I do it using the following vscode extention:
https://marketplace.visualstudio.com/items?itemName=webfreak.debug

my project root folder looks like the following:
-> myproject/
------> .vscode
----------> launch.json
------> app/
----------> try1.py
------> micropython/ 
----------> micropython_files
----------> ....

then i added 2 configurations in the launch.json:
```json
{
    "configurations": [
        { // this configuration runs micropython on a file that located in the app folder
             "type" : "gdb",
             "request" : "launch",
             "name" : "micropython",
             "target" : "ports/unix/micropython",
             "arguments" : "../app/try1.py",
             "cwd": "${workspaceRoot}/micropython/",
        },
        {
             "type" : "gdb",
             "request" : "launch",
             "name" : "mpy-cross",
             "target" : "mpy-cross",
             "arguments" : "../../app/try1.py",
             "cwd" : "${workspaceRoot}/micropython/mpy-cross"
        }
    ]
}
```

closed 2021-03-03 by ALEX010001