micropython/micropython #6954: Query: RP2040 short I2C writes: why length <= 2 instead of 0 ?

dhalbert, 2021-02-23:
@dpgeorge We also ran into the issue of not being able to do zero-length I2C writes on RP2040. I see your code uses bitbang I2C to substitute. But could you say why you are doing writes of length <= 2 instead of 0 by bitbang? I thought It was just address-only writes that were an issue, but I may be misreading something. Thanks for your insight.

dhalbert, 2021-03-11:
The RPi folks say there is no problem with 1 or 2-byte writes: https://github.com/raspberrypi/pico-sdk/issues/238.

dpgeorge, 2021-03-11:
> But could you say why you are doing writes of length <= 2 instead of 0 by bitbang?

I honestly don't remember!  It could be that at the time it was written the SDK also had issues with 1-byte writes and so I went conservative with 2 as the cutoff.  But really I don't remember.  If 1 and 2 byte writes are OK then it'd be good to change our code to use software only for 0-byte writes.

dpgeorge, 2022-07-12:
Hardware I2C now uses software for length=0 only, see a707fe50b085ec83722106609f6fd219faf9f030

closed 2022-07-12 by dpgeorge