micropython/micropython #6959: rp2: native C modules cause crash

peterhinch, 2021-02-24:
To replicate compile the `features0.c` demo for armv7m. The resultant `features0.mpy` imports OK, but any attempt to run it e.g.
```python
import features0
features0.factorial(3)
```
results in a crash.

kadamski, 2021-02-24:
Makes sense. RP Pico is Cortex-M0+ which is armv6m, not armv7m (cortex-m3). If you do a quick hack and modify py/dynruntime.mk to use `-mcpu=cortex-m0` instead of `-mcpu=cortex-m3` in case of `armv7m`, it works without a crash.

kadamski, 2021-02-24:
I was looking into fixing this problem properly but it is quite complicated, from what I've seen. But at least we know the root cause.

pigrew, 2021-08-21:
An example diff is posted [on the RPi forums](https://www.raspberrypi.org/forums/viewtopic.php?t=305744). It works for me.

jimmo, 2021-08-21:
See also https://github.com/micropython/micropython/issues/7616#issuecomment-894002358 for a related issue.

Fortunately this one is easier to work around!

dpgeorge, 2022-05-23:
Since c1b9d2259e77548053cba4a1f2cab726e6429a97, features0 and features1 natmod examples can build for `armv6m` and run on the rp2 port.

closed 2022-05-23 by dpgeorge