micropython/micropython #6961: rp2: Problem when creating more than 4 state machines

robert-hh, 2021-02-24:
I ran into a problem when I wanted to create more than 4 state machines. Five in may case. It seems that creating the fifth state machine interferes with the previous ones. Using the number sequence:
0 1 2 3 4 get trouble, even when SM 4 is not activated, just created.
The sequence 4 0 1 2 3 lets me at least state machines 0 - 3. It looks as if the creating SM 4 the instruction memory of PIO0 is used instead of that of PIO1
A quick glance at the code did not reveal an obvious place, and I do not have a simple test case yet. I will try to make one.

robert-hh, 2021-02-24:
I have no simple test case, so I extracted the code from the larger file. Calling test(0) does not set up SM 4, and this SM 1 is working and sending data. If I call test(1), SM 4 is set up, and SM 1 is silent.
```
from machine import Pin
import rp2

def test(do_sm4, data_pin=2, ctrl_pin=10):
# fmt: off

# define the PIO codes
# writing command
    @rp2.asm_pio(
        sideset_init=(rp2.PIO.OUT_HIGH,) * 3,
        out_init=(rp2.PIO.OUT_HIGH,) * 8,
        out_shiftdir=rp2.PIO.SHIFT_RIGHT,
        autopull=True,
        pull_thresh=8)
    def pio_cmd_write():
        out(pins, 8)            .side(0b100)
        nop()                   .side(0b110)

# just write byte data
    @rp2.asm_pio(
        sideset_init=(rp2.PIO.OUT_HIGH,) * 3,
        out_init=(rp2.PIO.OUT_HIGH,) * 8,
        out_shiftdir=rp2.PIO.SHIFT_RIGHT,
        autopull=True,
        pull_thresh=8)
    def pio_data_write_byte():
        out(pins, 8)            .side(0b101)
        nop()                   .side(0b111)

# just write pixel data
    @rp2.asm_pio(
        sideset_init=(rp2.PIO.OUT_HIGH,) * 3,
        out_init=(rp2.PIO.OUT_HIGH,) * 8,
        out_shiftdir=rp2.PIO.SHIFT_RIGHT,
        autopull=True,
        pull_thresh=24)
    def pio_data_write_triple():
        out(pins, 8)            .side(0b101)
        nop()                   .side(0b111)

# write 32 bit words, high order byte first

    @rp2.asm_pio(
        sideset_init=(rp2.PIO.OUT_HIGH,) * 3,
        out_init=(rp2.PIO.OUT_HIGH,) * 8,
        out_shiftdir=rp2.PIO.SHIFT_LEFT,
        autopull=True,
        pull_thresh=32)
    def pio_data_write_word():
        out(pins, 8)            .side(0b101)
        nop()                   .side(0b111)

# perform a data bus read, by first:
# - getting the number N of reads
# - emitting a command
# - setting the Pins to input mode
# - reading N bytes,with a longer delay at the first read 
# - setting back the pins to output mode
#
    @rp2.asm_pio(
        sideset_init=(rp2.PIO.OUT_HIGH,) * 3,
        out_init=(rp2.PIO.OUT_HIGH,) * 8,
        out_shiftdir=rp2.PIO.SHIFT_RIGHT,
        autopull=False,
        autopush=False,
        pull_thresh=16,
        push_thresh=8)
    def pio_cmd_data_read():
        pull()              .side(0b111)  # get number N - 1 of bytes to read
        mov(x, osr)         .side(0b111)  # save it to the counter
        pull()              .side(0b100)
        out(pins, 8)        .side(0b100)  # send the command
        nop()               .side(0b110)
        out(pindirs, 8)     .side(0b011) [3] # and switch to input mode
        nop()               .side(0b011) [3] # First read need a delay

        label("again")
        nop()               .side(0b111)
        in_(pins, 8)        .side(0b111)
        push()              .side(0b111)
        jmp(x_dec, "again") .side(0b011)

        pull()              .side(0b111)  # get the new pindir value
        out(pindirs, 8)     .side(0b111)  # and switch back to output mode

# fmt: on

    sm_data_write_byte = rp2.StateMachine(1, pio_data_write_byte, freq=25000000,
                        sideset_base=Pin(ctrl_pin), out_base=Pin(data_pin))
    sm_data_write_triple = rp2.StateMachine(0,pio_data_write_triple, freq=25000000,
                        sideset_base=Pin(ctrl_pin), out_base=Pin(data_pin))
    sm_cmd_write = rp2.StateMachine(3, pio_cmd_write, freq=25000000,
                        sideset_base=Pin(ctrl_pin), out_base=Pin(data_pin))
    sm_cmd_data_read = rp2.StateMachine(2, pio_cmd_data_read, freq=25000000,
                        sideset_base=Pin(ctrl_pin), out_base=Pin(data_pin),
                        in_base=Pin(data_pin))
    if do_sm4:
        sm_data_write_word = rp2.StateMachine(4, pio_data_write_word, freq=25000000,
                            sideset_base=Pin(ctrl_pin), out_base=Pin(data_pin))

    sm_data_write_byte.active(1)
    sm_data_write_byte.put(b"abcdefg")
    sm_data_write_byte.active(0)
```

robert-hh, 2021-02-25:
I found a reason for the stall: Programs from PIO0 and PIO1 must not use the same GPIO pins.
In my example that is the case. I did not find a hint in the documentation about it. But well, it might be there.

closed 2022-01-26 by robert-hh

cseatec, 2022-04-09:
Thank you for posting this topic!  I think there might be a bug here since not too many people perhaps use more than 4 statemachines, spread across both PIO banks 0 and 1.

I also now just ran into this same issue.  It would seem that any mix of statemachine assignments across both PIO0 and PIO1 results in a hang condition for all PIOs.  Specifically, I have 4 statemachines, each of which operate on a separate GPIO pins (which I even spread out based upon what I read in this topic, to no avail).  If I init and assign the state machines to IDs 0-3, everything works fine.  However, if I assign the 4 statemachines to say 0, 1, 2, and 4 (or anything between 4 and 7), the statemachines will no longer function.  I even tried making sure that the statemachines in PIO0 and the ones in PIO1 use separate assembly code segments (so that both PIOs weren't trying to gain access to the same code in the 32-inst memory bank.   That did not help either.  

I actually now need to use more than 4 statemachines in a project I am working on, so I need to somehow get past this issue. Does anyone know what is going on?