micropython/micropython #6964: git

rolandvs, 2021-02-25:
`lib/lwip` cannot be loaded as git reports error like:

```
git submodule update --init
fatal: Needed a single revision
Unable to find current revision in submodule path 'lib/lwip'
```

Any clue?

jimmo, 2021-02-25:
What does `git status` say?  (possibly mid-rebase or detached head?)

rolandvs, 2021-02-25:
I just tried a clean clone using the `git desktop` tool and after 20 minutes of creation time it failed.

I have attached the log file in compressed form. It shows near the end some 4 fatal errors.
I use the latest OSX on Intel machine as generic as possible.
[cloning_mpy.txt.zip](https://github.com/micropython/micropython/files/6042547/cloning_mpy.txt.zip)

It seems to concentrate around `lwip` and `tinyusb` as the other submodules always load.

At the end the log reports:

```
Cloning into '/Users/me/Documents/GitHub/micropython/lib/tinyusb/hw/mcu/raspberrypi/pico-sdk/lib/tinyusb/lib/lwip'...
fatal: unable to access 'https://git.savannah.nongnu.org/git/lwip.git/': The requested URL returned error: 502
fatal: clone of 'https://git.savannah.nongnu.org/git/lwip.git' into submodule path '/Users/me/Documents/GitHub/micropython/lib/tinyusb/hw/mcu/raspberrypi/pico-sdk/lib/tinyusb/lib/lwip' failed
Failed to clone 'lib/lwip'. Retry scheduled
...
Cloning into '/Users/me/Documents/GitHub/micropython/lib/tinyusb/hw/mcu/raspberrypi/pico-sdk/lib/tinyusb/lib/lwip'...
fatal: unable to access 'https://git.savannah.nongnu.org/git/lwip.git/': Failed to connect to git.savannah.nongnu.org port 443: Operation timed out
fatal: clone of 'https://git.savannah.nongnu.org/git/lwip.git' into submodule path '/Users/me/Documents/GitHub/micropython/lib/tinyusb/hw/mcu/raspberrypi/pico-sdk/lib/tinyusb/lib/lwip' failed
Failed to clone 'lib/lwip' a second time, aborting
Failed to recurse into submodule path 'lib/tinyusb/hw/mcu/raspberrypi/pico-sdk/lib/tinyusb'
Submodule path 'lib/tinyusb/hw/mcu/sony/cxd56/spresense-exported-sdk': checked out '2ec2a1538362696118dc3fdf56f33dacaf8f4067'
...
Submodule path 'lib/tinyusb/tools/uf2/hidapi': checked out 'a6a622ffb680c55da0de787ff93b80280498330f'
Failed to recurse into submodule path 'lib/tinyusb/hw/mcu/raspberrypi/pico-sdk'
Failed to recurse into submodule path 'lib/tinyusb'
```

rolandvs, 2021-02-25:
OK tried and tried again and now it works again so...

closed 2021-02-25 by rolandvs