micropython/micropython #6969: esp2866 restart couldn't stop

vidalouiswang, 2021-02-27:
`
def init():
    import net
    if net.connect('Vida', '3W9%5x.V2_4d,X5$'):
        print("ok")

init()
`
After running this code, esp8266 will reboot itself(hard reset) won't stop.

net.py:
`
import network
def connect(ssid,password=None):
    wlan = network.WLAN(network.STA_IF)
    wlan.active(True)
    import machine
    a = scan(ssid)
    if a:
        wlan.connect(ssid, password)
        while not wlan.isconnected():
            machine.idle()
        return True

def scan(ssid=None):
    wlan = network.WLAN(network.STA_IF)
    wlan.active(True)
    if ssid:
        a = wlan.scan()
        for i in a:
            if i[0] == bytes(ssid,"ascii"):
                return i
    else:
        return wlan.scan()
`
Need to erase and flash firmware in esptool will recover.

vidalouiswang, 2021-02-27:
Codes in net.py almost from offical examples.

vidalouiswang, 2021-02-27:
I also tried these simple code:
`
MicroPython v1.14 on 2021-02-02; ESP module with ESP8266

Type "help()" for more information.
>>> import network
>>> import machine
>>> wlan = network.WLAN(network.STA_IF)
>>> wlan.active(True)
#7 ets_task(4020ee68, 28, 3fff9e90, 10)
>>> wlan.connect("Vida", "3W9%5x.V2_4d,X5$")
>>> wlan.isconnected()
True
>>> ldld`��;�$�d�$`��s�l�l�l`��{�dǟ
`

Seems it will restart after connected to AP.

vidalouiswang, 2021-02-27:
> I also tried these simple code:
> '
> import network
> import machine
> wlan = network.WLAN(network.STA_IF)
> wlan.active(True)
> wlan.connect("ssid", "password")
> wlan.isconnected()
> `
> 
> Here is the result:
> 
> `MicroPython v1.14 on 2021-02-02; ESP module with ESP8266
> 
> Type "help()" for more information.
> 
> > > > import network
> > > > import machine
> > > > wlan = network.WLAN(network.STA_IF)
> > > > wlan.active(True)
> > > > #7 ets_task(4020ee68, 28, 3fff9e90, 10)
> > > > wlan.connect("ssid", "password")
> > > > wlan.isconnected()
> > > > True
> > > > ldld`���;�$�d���$`���s�l�l���l`���{�dǟ`
> 
> Seems it will restart after connected to AP.

Run same code, it recovered itself(won't restart again), I do nothing before that, only unplug in the board from the cable for about 2mins.

peterhinch, 2021-02-27:
I suggest you raise this support query in [the forum](http://forum.micropython.org/index.php) rather than in the MicroPython bug tracker.

vidalouiswang, 2021-02-27:
> I suggest you raise this support query in [the forum](http://forum.micropython.org/index.php) rather than in the MicroPython bug tracker.

Okay, thank you.

closed 2021-03-11 by dpgeorge