micropython/micropython #6970: Missing attributes

kdschlosser, 2021-02-27:
`memoryview` iis missing every single attribute/function

`c_contiguous`
`cast`
`contiguous`
`f_contiguous`
`format`
`hex`
`itemsize`
`nbytes`
`ndim`
`obj`
`readonly`
`release`
`shape`
`strides`
`suboffsets`
`tobytes`
`tolist`
`toreadonly`

`__iter__` is missing from `bytearray`, this one is strange because I am able to iterate over a bytearray. Not sure exactly how that is working..

also if I run the following code my ESP32 crashes and then boot loops

```python
for item in dir(memoryview):
    print(item)
```

but this code does work.

```python
test = bytearray()
for item in dir(memoryview(test)):
    print(item)
```

I am trying to get access to the underlying bytearray attached to the memory view object. II am doing this because I am using a custom `__iter__` method in order to add padding. I have another method that calculates the "width" which is the bit number where the last 1 is found across all bytes in the array.

I have sub classed the memory view object to do this

```python
class Character(memoryview):
    _pad = array.array('B', [0])
    _width = array.array('B', [0])

    @property
    def width(self):
        if self._width[0] == 0:
            for ii in range(8):
                for byte in self.obj:
                    if (byte >> ii) & 1:
                        break
                else:
                    break

            else:
                ii = 1

            self._width[0] = ii + 1
        return self._width[0]

    def pad(self, amount):
        self._pad[0] = amount
        return self

    def __iter__(self):
        res = bytearray([0] * (12 - len(self) + self._pad[0]))
        res += self
        res += bytearray([0] * self._pad[0])
        return iter(memoryview(res))

```

if you look in the `__iter__` method you will notice how I am adding the contents of the memoryview object to a bytearray, strangely this works. but in the width property if I use `for byte in self:` it is going to use the `__iter__` method I created. Doing that does work but the program then has to iterate over bytes that are 0's because of the padding being added. Now I know that I can do the following to help sped things up

```python
for byte in self:
    if byte = 0:
        continue
```

I shouldn't have to do this, there should be access to the underlying object

I have another question. In CPython if I want to speed up a for loop I am able to do this using list comprehension. Even if I do not set the list into a variable it is going to speed p the processing some 200 times.

```python
def _do():
    # put some code here

[_do() for _ in range(10000000)]
```

the code above is going to run leaps and bounds faster then the code below.

```python
def _do():
    # put some code here

for _ in range(10000000):
    _do()
```

is this also the case with MicroPython? I find this to be one of best (if not the best) thing to do when optimizing python code. It makes a huge improvement in the speed in which a program runs if having to iterate over large data sets or even when iterating over smaller data sets often.

stinos, 2021-02-28:
> memoryview is missing every single attribute/function

> __iter__ is missing from bytearray, this one is strange because I am able to iterate over a bytearray. Not sure exactly how that is working..

It's not unlikely some of those are actually not missing, but just not discovered by `dir` or `__dict__` or similar. Main reason being: this is not CPython and not everything is implemented in a CPython-compatible way. So for example a bytearray can be iterated because it supports that using the mechanims used internally for iteration, but because this is 'micro' Python there is no `__iter__`  because that would require extra code meaning larger executables meanning less room left on microcontrollers for user programs. While at the same time the benefot of exposing it is rather small.

> the code above is going to run leaps and bounds faster then the code below

Can you show actual code for which this is the case? Asking because this is rather unlikely (and I cannot reproduce it in any way). Even ruling out that it would depend on what `_do` actually does, it makes little sense, If anything they should run in about the same time. It's more likely this is a measurement error of some kind.

kdschlosser, 2021-03-01:
```python
import timeit

MILLION_NUMBERS = list(range(1000000))

def _do():
    output = []
    for element in MILLION_NUMBERS:
        if not element % 2:
            output.append(element)
    return output

def _do2():
    return [number for number in MILLION_NUMBERS if not number % 2]

print('for loop:', timeit.timeit('_do()', number=10, globals=globals()) / 10)
print('list comprehension:', timeit.timeit('_do2()', number=10, globals=globals()) / 10)
```

run the script and see the output. the output is the average time it takes to run each of the tests. each test is run 10 times..
output
```
for loop: 0.1545252838
list comprehension: 0.12150151189999998
```

kdschlosser, 2021-03-01:
and it is not a measurement error, It is a well known and documented thing about Python. Just do a google search for "python list comprehension faster than for loop"

stinos, 2021-03-01:
Sure, but this example actually creates a list (of which it is indeed known that comprehension is usually faster), your original just loops, that's a big difference :)

kdschlosser, 2021-03-01:
Is there a way in MicroPython to access the underlying object of a memoryview instance?

kdschlosser, 2021-03-01:
oh for cripes sake. It was an example.  What iis the intention of questioning it if you already knew that list comprehensions are indeed faster? what purpose did it serve to make me waste something I am never going to get back??? (TIME).

You as of yet have not answered a single question I have asked. so what is your intention other then filling this issue with nothing constructive or helpful?

stinos, 2021-03-01:
> It was an example. What iis the intention of questioning it if you already knew that list comprehensions are indeed faster? 

Because that particular example plus your explanation, was something else than what I actually know, so I asked for more information, to verify that perhaps there are things I don't know. No reason to get upset about that.

> You as of yet have not answered a single question I have asked

I did ask questions though, to avoid providing answers which makes no sense, plus I explained why all those attributes are missing which seems quite helpful to me. Sorry if you did not understand it that way.

closed 2025-03-03 by kdschlosser