micropython/micropython #6971: bare-arm is missing include flags for user modules

KaiJan57, 2021-02-28:
https://github.com/micropython/micropython/blob/cdaec0dcaffbfcb58e190738cf6e9d34541464f0/ports/bare-arm/Makefile#L19
When trying to compile bare-arm with user modules, include flags are missing. It took me hours to find that this line is missing `$(CFLAGS_MOD)`, so maybe add that by default? I believe that could spare others many hours of frustration!

dpgeorge, 2021-03-11:
The bare-arm (and minimal) ports are not really intended to be built using user C modules.  And it's not enough to just add `$(CFLAGS_MOD)`, it also needs additions for `$(CFLAGS_EXTRA)`, `$(SRC_MOD)` and `$(LDFLAGS_MOD)`, as well as `$(SRC_QSTR)`.  And then more stuff for C++ support (which is needed to build the default set of user C modules in the examples/ directory).

closed 2021-03-16 by KaiJan57

dpgeorge, 2021-03-21:
See 4fc2866f45692b05829d71b636724e690b786307 for improvements to the bare-arm port.