micropython/micropython #6974: machine.SPI initialisation can fail silently

peterhinch, 2021-03-01:
This has caused confusion with a number of users, notably on ESP32, for example [here](https://github.com/peterhinch/micropython-nano-gui/issues/6). Given this line
```python
spi = SPI(1, 10000000)
```
the docs give a reasonable expectation of an SPI bus on the default pins (12, 13, 14). In practice the bus silently fails to work. It seems necessary explicitly to specify these pins even though they are the default:
```python
spi = SPI(1, 10_000_000, sck=Pin(14), mosi=Pin(13), miso=Pin(12))
```
If pins are not specified the code should either throw an exception or use the default pins.

dpgeorge, 2021-03-11:
Thanks for the report.  See #7011 for a fix for this.

dpgeorge, 2021-03-12:
Should be fixed by b24fcd7aec4b34064e9d9016d417d4ca3cc925cc

closed 2021-03-12 by dpgeorge