micropython/micropython #6975: Is there a async websocket client module can use?

vidalouiswang, 2021-03-01:
I didn't find any info in offical docs, so I write a simple websocket client.
But I it will stuck main thread if I want to keep looking the data from socket,
I tried uasyncio.open_connection and any method I could find, nothing useful.
Now, if I want to get data from websocket server, the main code will not run, if I want to run main code, the socket won't push any data.
Here is the code(isn't finished):

`
class WebSocket():
    def __init__(a,b=None):
        d = self
        d.socket = None
        d.host = a
        b = b or {}
        keys = b.keys()
        def isIn(key):
            if key in keys:
                return b[key]
            else:
                return False
        d.port = isIn("port") or 80
        d.protocolVersion = isIn("protocolVersion") or 13
        d.origin = isIn("origin") or "Cloud Player"
        d.keepAlive = 1e3 * (isIn("keepAlive") or 6e4)
        d.masking = isIn("masking") or True
        d.path = isIn("path") or "/"
        d.protocol = isIn("protocol") or ""
        d.lastData = ""
        d.events = []

        try:
            import binascii as asc
        except:
            import ubinascii as asc

        def r():
            import uos as os
            a = ""
            while len(a) < 16:
                a += str(os.urandom(1)[0])
            a = a[0:16]
            return asc.b2a_base64(a).decode()[:-1]
        import uhashlib as lib
        c = r()
        f = c + "258EAFA5-E914-47DA-95CA-C5AB0DC85B11"
        d.key = {"source": c, "hashed": asc.b2a_base64(lib.sha1(f).digest()).decode()[:-1]}

        d.dbl = chr(13) + chr(10)
        self.initialize()

    def initialize(self):
        import usocket
        import uselect
        self.socket = usocket.socket()
        import uasyncio as io

        reader,writer = io.run(io.open_connection(usocket.getaddrinfo(self.host, self.port)[0][-1][0],self.port))
        self.writer = writer
        self.reader = reader
        self.handshake()
        
    def read(self):
        import uasyncio as io
        if not self.reader:
            print("empty reader")
        else:
            data = io.run(self.reader.read(512))
            if data:
                print(data)
            else:
                print("empty")
        
        
    def write(self,data):
        import uasyncio as io
        if not self.writer:
            print("empty writer")
        else:
            data = bytes(data,"utf8")
            print(data)
            self.writer.write(data)
            io.run(self.writer.drain())
        
    
    def handshake(self):
        h = self
        rn = h.dbl
        res = "GET " + h.path + " HTTP/1.1" + rn
        res += "Host: " + h.host + rn
        res += "Upgrade: websocket" + rn
        res += "Connection: Upgrade" + rn
        res += "Sec-WebSocket-Key: " + h.key['source'] + rn
        res += "Sec-WebSocket-Version: " + str(h.protocolVersion) + rn
        res += "Origin: " + h.origin
        res += rn + rn
        self.write(res)
        import machine
        while True:
            self.read()
            machine.idle()
`

straga, 2021-03-03:
Run asyncio in the thread. I do that.

vidalouiswang, 2021-03-03:
> Run asyncio in the thread. I do that.

On esp8266? I didn't find that module in micropython.

straga, 2021-03-03:
ESP32.

marcidy, 2021-03-09:
I have a working example: https://github.com/marcidy/micropython-uasyncio-webexample

No ssl yet, waiting on the uasyncio + ssl and will update the example then.

vidalouiswang, 2021-03-12:
> I have a working example: https://github.com/marcidy/micropython-uasyncio-webexample
> 
> No ssl yet, waiting on the uasyncio + ssl and will update the example then.

Thank you.

closed 2021-03-12 by vidalouiswang

beyonlo, 2022-01-14:
Hello @marcidy 

> I have a working example: https://github.com/marcidy/micropython-uasyncio-webexample
> 
> No ssl yet, waiting on the uasyncio + ssl and will update the example then.

I think that uasyncio + ssl is working now, right?

I would like to run the uasyncio websocket server on ESP32.

Thank you in advance!

marcidy, 2022-01-17:
@beyonlo I don't see ssl support in uasyncio in 1.17,  open_connection opens a socket and doesn't wrap it

beyonlo, 2022-01-18:
@marcidy yes, 1.17 has SSL support. Look about the ussl module. A HTTP Server with SSL for example, like this official example, works https://github.com/micropython/micropython/blob/master/examples/network/http_server_ssl.py

And now, we are in the 1.18. Released yesterday!