micropython/micropython #6977: Wasted Memory?

kdschlosser, 2021-03-02:
I am going to be using an ESP32 and the program I am running on it is going to produce some memory challenges for me. I have a need for using the framebuf and from looking at the code there is 1016  bytes of ram that gets consumed for something that may not get used. The memory gets used by the "built in" font, the array that holds the font has 127 elements and each of those elements uses 8 bytes of data. that's a total of 1016 Not everyone that uses the framebuf class uses the font but they have to take a 1016 byte memory hit. There is no way to turn on or off the loading of this array not even with a preprocessor macro. It would be nice if there was a way to do this without the need to compile the firmware. 1K of memory is a pretty sizeable amount when dealing with the low availability of memory on some of these micro controllers. it would also be beneficial to have the array declared inside of the framebuf_text function. This would allow the memory to get reclaimed during a GC once the function has finished running.

dpgeorge, 2021-03-11:
I'm pretty sure the font data is stored in ROM/flash.  Did you see otherwise?

kdschlosser, 2021-03-11:
when importing the framebuf module the font gets loaded into memory yes?

I am trying to get an understanding of how memory is allocated. because it appears that the memory gets allocated for an object when the module gets loaded and not when the code for creating an object actually gets run.

if you put the following code into a file called test.py and load it onto an ESP32 and from the main.py file import it. There is a massive amount of ram that gets used even tho the function has never been called.

```python
def test():
    variable = bytearray([0xFFFFFFFFFFFFFFFF] * 5000)
```

Here is another example, this one I believe i a memory leak because the previously used memory doesn't get release when doing a GC

```python
test = bytearray([0xFFFFFFFFFFFFFFFF] * 256)

while len(test) > 1:
    test = test[:-1]
```

in the last example if I perform a GC juust after the bytearray gets declared and print out the memory use and then do a GC after the while loop and print out the memory use I should see a drop of 2040 bytes of memory. This doesn't happen.

If I have a module that has 600 lines of code in it I perform a GC and print out the memory use then import the module. at the top of the module I perform yet another GC and print out the memory use and there is 14,000 bytes of memory that is gone. No code other then running a GC and printing out the memory use has taken place. so why is all of this memory being allocated for code that has not run yet?

I know that building a custom firmware and compiling my modules into byte code uses less ram.. Unfortunately  this does not seem like it is possible to do on Windows, that is what the documentation states.

kdschlosser, 2021-03-11:
I also have another question. How come modules that are baked into the firmware are able to be loaded into ROM/Flash instead of using large amounts of RAM and this is not done with .py files? Or how come available space in ROM/Flash cannot be added to the available RAM?

I have tried all kinds of things to reduce memory use and I almost always end up using more.

and example is if I have a class that has say 16 boolean attributes. each of these attributes is a Python integer and is going to consume I believe at at least 4 bytes just for the data. So what I did was create functions to do some bit shifting on a single python 16 bit integer and store the boolean value as a bit value instead. declaring 16 functions uses up way more memory then 16 integers.

dpgeorge, 2021-03-11:
> when importing the framebuf module the font gets loaded into memory yes?

No.  The font data stays in ROM/flash.  Eg:
```
MicroPython v1.14-90-gaf0cae9f8 on 2021-03-11; PYBD-SF6W with STM32F767IIK
Type "help()" for more information.
>>> import micropython
>>> micropython.mem_info(1)
stack: 500 out of 23552
GC: total: 391936, used: 944, free: 390992
 No. of 1-blocks: 11, 2-blocks: 7, max blk sz: 18, max free sz: 24425
GC memory layout; from 2001a4f0:
00000: h=hhhhh=================hh=======h=======h=hhh=h=BhBhh=.......h=
00400: .....h=.........................................................
       (380 lines all free)
5f800: ................................................
>>> import framebuf
>>> micropython.mem_info(1)
stack: 500 out of 23552
GC: total: 391936, used: 1136, free: 390800
 No. of 1-blocks: 15, 2-blocks: 11, max blk sz: 18, max free sz: 24414
GC memory layout; from 2001a4f0:
00000: h=hhhhh=================hh=======h=======h=hhh=h=BhBhh=hBh=hh=h=
00400: B....h=..h=.....h=..............................................
       (380 lines all free)
5f800: ................................................
>>> 
```
So importing framebuf takes only 1136-944 = 192 bytes (probably less, some of that memory will be reclaimed on next GC).

peterhinch, 2021-03-12:
@kdschlosser Regarding your example, compare the two ways of creating your 1-byte bytearray.

Successive reduction.
```python
>>> import gc
>>> gc.collect()
>>> gc.mem_free()
101504
>>> test = bytearray([0xFFFFFFFFFFFFFFFF] * 256)
>>> gc.collect()
>>> gc.mem_free()
101072
>>> 
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== while len(test) > 1:
===     test = test[:-1]
=== 
>>> gc.collect()
>>> gc.mem_free()
101280
>>> 
```
Direct creation.
```python
>>> import gc
>>> gc.collect()
>>> gc.mem_free()
101504
>>> test = bytearray(b'\xff')
>>> gc.collect()
>>> gc.mem_free()
101360
```
The difference is only 80 bytes. Incidentally queries like this are best raised in [the forum](http://forum.micropython.org/index.php).

kdschlosser, 2021-03-13:
@peterhinch 

> Incidentally queries like this are best raised in the forum.

I did.

101504 - 101072 = 432 bytes used for an array that has 256 8 byte integers???
256 * 8 = 2048

OK so the memory is getting allocated before the bytearray even gets declared..

101280 - 101072 = 208

and when removing 255 of the 8 byte integers only 208 bytes get freed??? the rest of the used memory simply goes into limbo and cannot be used again for anything else?

I am trying to understand how this is not considered a memory leak.

because if this gets done even more memory ends up getting consumed before the code actually runs.

```python
import gc
gc.collect()
print(gc.mem_free())
test = bytearray([0xFFFFFFFFFFFFFFFF] * 256)
gc.collect()
print(gc.mem_free())
while len(test) > 1:
    test = test[:-1]
gc.collect()
print(gc.mem_free())
test = bytearray([0xFFFFFFFFFFFFFFFF] * 512)
gc.collect()
print(gc.mem_free())
while len(test) > 1:
    test = test[:-1]
gc.collect()
print(gc.mem_free())
```

and this is the output.
108544
108272
108000
107504
108000

The amount of free memory when the module loads (108544) is lower with the second bytearray deceleration then when I only declare it a single time.

There is only 496 bytes consumed for an array that should use 4096 bytes of memory.

That 496 bytes does get released when shrinking the second byte array. unlike the first byte array that has 492 bytes that doesn't get released not including the rest of the bytes that get consumed before the creation of the byte array even takes place.

This test is run when the above code placed into a module and that module gets imported. I would expect the results from using the REPL to be different.

kdschlosser, 2021-03-13:
now if I add this code to main.py 

```python
import gc

gc.collect()
print(gc.mem_free())

test = bytearray([0xFFFFFFFFFFFFFFFF] * 1024)
gc.collect()
print(gc.mem_free())
```

this is the output
```
109984
104832
```

and then when I put this code into the main.py file

```python
import gc

gc.collect()
print(gc.mem_free())

test = bytearray([0xFFFFFFFFFFFFFFFF] * 1024)
gc.collect()
print(gc.mem_free())

test = bytearray([0xFFFFFFFFFFFFFFFF] * 1024)
test = bytearray([0xFFFFFFFFFFFFFFFF] * 1024)
test = bytearray([0xFFFFFFFFFFFFFFFF] * 1024)
test = bytearray([0xFFFFFFFFFFFFFFFF] * 1024)
test = bytearray([0xFFFFFFFFFFFFFFFF] * 1024)

gc.collect()
print(gc.mem_free())
```

this is the outpuut
```
109616
104464
103440
```

This is has got to be a memory leak. Look at the ram consumption between the 2 examples when the module loads.
first example 109984
second example 109616

368 bytes get used for something, I do not have a clue as to what
This is the even more important part.  I am declaring the exact same variable with the exact same bytearray that is filled with  1024 8 byte integers. I do this 5 times after the initial declaration.

free memory after the first declaration: 104464
free memory after the next 5 declarations: 103440

1024 bytes used and cannot be garbage collected.

Something has to be holding a reference to something when it shouldn't be. This is a memory leak! and I still do not fully grasp why memory is being allocated at module import for code that has not even run yet. and may never run depending on the program. 368 bytes getting used when importing a module if the declaration of a variable takes place several times? 

That means that code like this is going to chomp up memory that cannot be released.
```python
some_data_1 = [10] * 10
some_data_2 = [30] * 30
some_data_3 = [20] * 20

index = 0
while index < 10:
    print(some_data_1[index]
    index += 1

index = 0
while index < 30:
    print(some_data_2[index]
    index += 1

index = 0
while index < 20:
    print(some_data_3[index]
    index += 1
```

kdschlosser, 2021-03-13:
I'm a ding bat.. byte array is only able to story bytes and not the 8 byte unsigned long long. so here is the ouuutputt uusuing array.array instead.

```
109952
97616
```
109952 - 97616 = 12336
12336 bytes used for an array holding 1024 8 byte integers. seems to be pretty high.

Now here is the real damage. declaring the save variable 5 more times.
```
109568
97232
72624
```

now only is there a 384 byte hit at the import of the module that cannot be gc'd,  there is 24608 bytes that get used for declaring the same variable 5 more times that cannot be gc'd either.

```python
import gc
import array
gc.collect()
print(gc.mem_free())

test = array.array('Q', [0xFFFFFFFFFFFFFFFF] * 1024)
gc.collect()
print(gc.mem_free())

test = array.array('Q', [0xFFFFFFFFFFFFFFFF] * 1024)
test = array.array('Q', [0xFFFFFFFFFFFFFFFF] * 1024)
test = array.array('Q', [0xFFFFFFFFFFFFFFFF] * 1024)
test = array.array('Q', [0xFFFFFFFFFFFFFFFF] * 1024)
test = array.array('Q', [0xFFFFFFFFFFFFFFFF] * 1024)

gc.collect()
print(gc.mem_free())
```

tve, 2021-03-13:
I didn't follow all the twists and turns of your examples, but what comes to mind:
- something like `[0xFFFFFFFFFFFFFFFF] * 1024` maybe a constant expression that gets evaluated at parse time, thus the result needs to be stored somewhere
- your variables like `test` or `some_data_x` are module-global, so need to remain allocated as long as the module exists, which is forever
- in your example where you assign `test = ...` 5 times, doesn't gc.collect() get called as MPY is parsing the file? Thus the extra bytes would be the result of compiling the following statements?

kdschlosser, 2021-03-13:
This right here

```python
test = array.array('Q', [0xFFFFFFFFFFFFFFFF] * 1024)
test = array.array('Q', [0xFFFFFFFFFFFFFFFF] * 1024)
test = array.array('Q', [0xFFFFFFFFFFFFFFFF] * 1024)
test = array.array('Q', [0xFFFFFFFFFFFFFFFF] * 1024)
test = array.array('Q', [0xFFFFFFFFFFFFFFFF] * 1024)
```

uses up 24608 bytes of memory. This should not be happening. I am declaring the same variable over and over again. so as each declaration gets run the one before it should get released and garbage collected. This is not happening. 

my example with the "some_data" was not about the some_data variable but the index variable that gets declared more then once. It shows a real world scenario where that can take place. and as seen with the above code the memory is not getting released and it cannot get garbage collected. 

I tested this some more by making a class and creating an instance attribute.
If I re declare the instance attribute from the module level there is a memory leak
If I re declare it in an instance's method and I call that method 5 times, memory leak. less severe than above

running this code 
```python
import gc
import array
gc.collect()
print(gc.mem_free())

class Test(object):

    def __init__(self):
        self.test = array.array('Q', [0xFFFFFFFFFFFFFFFF] * 1024)

        gc.collect()
        print(gc.mem_free())

    def do(self):
        self.test = array.array('Q', [0xFFFFFFFFFFFFFFFF] * 1024)

test = Test()

test.do()
test.do()
test.do()
test.do()
test.do()

gc.collect()
print(gc.mem_free())
```

outputs this
```
109600
97072
92976
```
which is a leak.

if I run this code
```python
import gc
import array
gc.collect()
print(gc.mem_free())

class Test(object):

    def __init__(self):
        self.test = array.array('Q', [0xFFFFFFFFFFFFFFFF] * 1024)

        gc.collect()
        print(gc.mem_free())

test = Test()

test.test = array.array('Q', [0xFFFFFFFFFFFFFFFF] * 1024)
test.test = array.array('Q', [0xFFFFFFFFFFFFFFFF] * 1024)
test.test = array.array('Q', [0xFFFFFFFFFFFFFFFF] * 1024)
test.test = array.array('Q', [0xFFFFFFFFFFFFFFFF] * 1024)
test.test = array.array('Q', [0xFFFFFFFFFFFFFFFF] * 1024)

gc.collect()
print(gc.mem_free())
```

I get this as the output
```
109408
96896
84608
```
larger memory leak

and if I run this code
```python
import gc
import array
gc.collect()
print(gc.mem_free())

class Test(object):

    def __init__(self):
        self.test = array.array('Q', [0xFFFFFFFFFFFFFFFF] * 1024)

        gc.collect()
        print(gc.mem_free())

test = Test()

for _ in range(5):
    test.test = array.array('Q', [0xFFFFFFFFFFFFFFFF] * 1024)

gc.collect()
print(gc.mem_free())
``` 

this is the output
```python
109696
97184
72576
```

and the leak is even larger

kdschlosser, 2021-03-13:
ok so when I declare a variable that already exists and there is no reference to the old one that is being held other then whatever mechanics handle it internally in Micropython when I "write over" it there should no longer be a reference to it held anywhere and it should then be able to be garbage collected. Something in MicroPython is holding reference to a variable that has been written over and it shouldn't be

robert-hh, 2021-03-13:
If I run the following little script:
```
import gc
import array
import time
gc.collect()
print(gc.mem_free())

while True:
    test = array.array('Q', [0xFFFFFFFFFFFFFFFF] * 1024)
    gc.collect()
    print(gc.mem_free())
    time.sleep(1)
```
That's what I get:
```
106032
93696
89600
85472
85472
85472
85472
85472
85472
89600
85504
85504
81376
85504
85504
81376
81376
85472
85472
81344
77216
89600
89600
85472
```
That's a stable behavior. It has some dips, because the print statement sometimes gets memory used by the big array, and therefore not the previous space can be re-used. But that sorts out eventually.

tve, 2021-03-13:
If run a file with gc.collect() at the start and another gc.collect() at the end you will measure the size of the compiled code between the two gc.collect calls.

robert-hh, 2024-08-18:
Similar topics are regularly discussed if people get aware of  the dynamic memory state.

closed 2024-08-18 by robert-hh