micropython/micropython #6979: Trying to add the example C modules to the compiler

Krasn4ck, 2021-03-02:
Hi again.
I have been trying to compile the default C modules, but at the moment of generating the .hex file I get the following error:

```
LINK build-OPHYRA/firmware.elf
/bin/sh: -c: line 0: syntax error near unexpected token `('
/bin/sh: -c: line 0: `arm-none-eabi-ld -nostdlib -L boards -Tboards/stm32f407.ld -Tboards/common_ifs.ld -Map=build-OPHYRA/firmware.map --cref --defsym=_estack_reserve=8 --gc-sections -Lc:/ ./ (x86)/ ./ ./ toolchain/ 2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard/  -o build-OPHYRA/firmware.elf build-OPHYRA/py/mpstate.o build-OPHYRA/py/nlr.o build-OPHYRA/py/nlrx86.o build-OPHYRA/py/nlrx64.o build-OPHYRA/py/nlrthumb.o build-OPHYRA/py/nlrpowerpc.o build-OPHYRA/py/nlrxtensa.o build-OPHYRA/py/nlrsetjmp.o build-OPHYRA/py/malloc.o build-OPHYRA/py/gc.o build-OPHYRA/py/pystack.o build-OPHYRA/py/qstr.o build-OPHYRA/py/vstr.o build-OPHYRA/py/mpprint.o build-OPHYRA/py/unicode.o build-OPHYRA/py/mpz.o build-OPHYRA/py/reader.o build-OPHYRA/py/lexer.o build-OPHYRA/py/parse.o build-OPHYRA/py/scope.o build-OPHYRA/py/compile.o build-OPHYRA/py/emitcommon.o build-OPHYRA/py/emitbc.o build-OPHYRA/py/asmbase.o build-OPHYRA/py/asmx64.o build-OPHYRA/py/emitnx64.o build-OPHYRA/py/asmx86.o build-OPHYRA/py/emitnx86.o build-OPHYRA/py/asmthumb.o build-OPHYRA/py/emitnthumb.o build-OPHYRA/py/emitinlinethumb.o build-OPHYRA/py/asmarm.o build-OPHYRA/py/emitnarm.o build-OPHYRA/py/asmxtensa.o build-OPHYRA/py/emitnxtensa.o build-OPHYRA/py/emitinlinextensa.o build-OPHYRA/py/emitnxtensawin.o build-OPHYRA/py/formatfloat.o build-OPHYRA/py/parsenumbase.o build-OPHYRA/py/parsenum.o build-OPHYRA/py/emitglue.o build-OPHYRA/py/persistentcode.o build-OPHYRA/py/runtime.o build-OPHYRA/py/runtime_utils.o build-OPHYRA/py/scheduler.o build-OPHYRA/py/nativeglue.o build-OPHYRA/py/pairheap.o build-OPHYRA/py/ringbuf.o build-OPHYRA/py/stackctrl.o build-OPHYRA/py/argcheck.o build-OPHYRA/py/warning.o build-OPHYRA/py/profile.o build-OPHYRA/py/map.o build-OPHYRA/py/obj.o build-OPHYRA/py/objarray.o build-OPHYRA/py/objattrtuple.o build-OPHYRA/py/objbool.o build-OPHYRA/py/objboundmeth.o build-OPHYRA/py/objcell.o build-OPHYRA/py/objclosure.o build-OPHYRA/py/objcomplex.o build-OPHYRA/py/objdeque.o build-OPHYRA/py/objdict.o build-OPHYRA/py/objenumerate.o build-OPHYRA/py/objexcept.o build-OPHYRA/py/objfilter.o build-OPHYRA/py/objfloat.o build-OPHYRA/py/objfun.o build-OPHYRA/py/objgenerator.o build-OPHYRA/py/objgetitemiter.o build-OPHYRA/py/objint.o build-OPHYRA/py/objint_longlong.o build-OPHYRA/py/objint_mpz.o build-OPHYRA/py/objlist.o build-OPHYRA/py/objmap.o build-OPHYRA/py/objmodule.o build-OPHYRA/py/objobject.o build-OPHYRA/py/objpolyiter.o build-OPHYRA/py/objproperty.o build-OPHYRA/py/objnone.o build-OPHYRA/py/objnamedtuple.o build-OPHYRA/py/objrange.o build-OPHYRA/py/objreversed.o build-OPHYRA/py/objset.o build-OPHYRA/py/objsingleton.o build-OPHYRA/py/objslice.o build-OPHYRA/py/objstr.o build-OPHYRA/py/objstrunicode.o build-OPHYRA/py/objstringio.o build-OPHYRA/py/objtuple.o build-OPHYRA/py/objtype.o build-OPHYRA/py/objzip.o build-OPHYRA/py/opmethods.o build-OPHYRA/py/sequence.o build-OPHYRA/py/stream.o build-OPHYRA/py/binary.o build-OPHYRA/py/builtinimport.o build-OPHYRA/py/builtinevex.o build-OPHYRA/py/builtinhelp.o build-OPHYRA/py/modarray.o build-OPHYRA/py/modbuiltins.o build-OPHYRA/py/modcollections.o build-OPHYRA/py/modgc.o build-OPHYRA/py/modio.o build-OPHYRA/py/modmath.o build-OPHYRA/py/modcmath.o build-OPHYRA/py/modmicropython.o build-OPHYRA/py/modstruct.o build-OPHYRA/py/modsys.o build-OPHYRA/py/moduerrno.o build-OPHYRA/py/modthread.o build-OPHYRA/py/vm.o build-OPHYRA/py/bc.o build-OPHYRA/py/showbc.o build-OPHYRA/py/repl.o build-OPHYRA/py/smallint.o build-OPHYRA/py/frozenmod.o build-OPHYRA/extmod/moduasyncio.o build-OPHYRA/extmod/moductypes.o build-OPHYRA/extmod/modujson.o build-OPHYRA/extmod/modure.o build-OPHYRA/extmod/moduzlib.o build-OPHYRA/extmod/moduheapq.o build-OPHYRA/extmod/modutimeq.o build-OPHYRA/extmod/moduhashlib.o build-OPHYRA/extmod/moducryptolib.o build-OPHYRA/extmod/modubinascii.o build-OPHYRA/extmod/virtpin.o build-OPHYRA/extmod/machine_mem.o build-OPHYRA/extmod/machine_pinbase.o build-OPHYRA/extmod/machine_signal.o build-OPHYRA/extmod/machine_pulse.o build-OPHYRA/extmod/machine_i2c.o build-OPHYRA/extmod/machine_spi.o build-OPHYRA/extmod/modbluetooth.o build-OPHYRA/extmod/modussl_axtls.o build-OPHYRA/extmod/modussl_mbedtls.o build-OPHYRA/extmod/modurandom.o build-OPHYRA/extmod/moduselect.o build-OPHYRA/extmod/moduwebsocket.o build-OPHYRA/extmod/modwebrepl.o build-OPHYRA/extmod/modframebuf.o build-OPHYRA/extmod/vfs.o build-OPHYRA/extmod/vfs_blockdev.o build-OPHYRA/extmod/vfs_reader.o build-OPHYRA/extmod/vfs_posix.o build-OPHYRA/extmod/vfs_posix_file.o build-OPHYRA/extmod/vfs_fat.o build-OPHYRA/extmod/vfs_fat_diskio.o build-OPHYRA/extmod/vfs_fat_file.o build-OPHYRA/extmod/vfs_lfs.o build-OPHYRA/extmod/utime_mphal.o build-OPHYRA/extmod/uos_dupterm.o build-OPHYRA/lib/embed/abort_.o build-OPHYRA/lib/utils/printf.o build-OPHYRA/build-OPHYRA/frozen_content.o build-OPHYRA/lib/libc/string0.o build-OPHYRA/lib/mp-readline/readline.o build-OPHYRA/lib/netutils/netutils.o build-OPHYRA/lib/netutils/trace.o build-OPHYRA/lib/netutils/dhcpserver.o build-OPHYRA/lib/timeutils/timeutils.o build-OPHYRA/lib/utils/gchelper_native.o build-OPHYRA/lib/utils/pyexec.o build-OPHYRA/lib/utils/interrupt_char.o build-OPHYRA/lib/utils/sys_stdio_mphal.o build-OPHYRA/lib/utils/mpirq.o build-OPHYRA/lib/libm/math.o build-OPHYRA/lib/libm/acoshf.o build-OPHYRA/lib/libm/asinfacosf.o build-OPHYRA/lib/libm/asinhf.o build-OPHYRA/lib/libm/atan2f.o build-OPHYRA/lib/libm/atanf.o build-OPHYRA/lib/libm/atanhf.o build-OPHYRA/lib/libm/ef_rem_pio2.o build-OPHYRA/lib/libm/erf_lgamma.o build-OPHYRA/lib/libm/fmodf.o build-OPHYRA/lib/libm/kf_cos.o build-OPHYRA/lib/libm/kf_rem_pio2.o build-OPHYRA/lib/libm/kf_sin.o build-OPHYRA/lib/libm/kf_tan.o build-OPHYRA/lib/libm/log1pf.o build-OPHYRA/lib/libm/nearbyintf.o build-OPHYRA/lib/libm/roundf.o build-OPHYRA/lib/libm/sf_cos.o build-OPHYRA/lib/libm/sf_erf.o build-OPHYRA/lib/libm/sf_frexp.o build-OPHYRA/lib/libm/sf_ldexp.o build-OPHYRA/lib/libm/sf_modf.o build-OPHYRA/lib/libm/sf_sin.o build-OPHYRA/lib/libm/sf_tan.o build-OPHYRA/lib/libm/wf_lgamma.o build-OPHYRA/lib/libm/wf_tgamma.o build-OPHYRA/lib/libm/thumb_vfp_sqrtf.o build-OPHYRA/extmod/modonewire.o build-OPHYRA/drivers/bus/softspi.o build-OPHYRA/drivers/bus/softqspi.o build-OPHYRA/drivers/memory/spiflash.o build-OPHYRA/drivers/dht/dht.o build-OPHYRA/lib/stm32lib/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal.o build-OPHYRA/lib/stm32lib/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_adc.o build-OPHYRA/lib/stm32lib/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_adc_ex.o build-OPHYRA/lib/stm32lib/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_cortex.o build-OPHYRA/lib/stm32lib/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_dma.o build-OPHYRA/lib/stm32lib/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_flash.o build-OPHYRA/lib/stm32lib/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_flash_ex.o build-OPHYRA/lib/stm32lib/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_gpio.o build-OPHYRA/lib/stm32lib/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_i2c.o build-OPHYRA/lib/stm32lib/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_pcd.o build-OPHYRA/lib/stm32lib/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_pcd_ex.o build-OPHYRA/lib/stm32lib/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_pwr.o build-OPHYRA/lib/stm32lib/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_pwr_ex.o build-OPHYRA/lib/stm32lib/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_rcc.o build-OPHYRA/lib/stm32lib/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_rcc_ex.o build-OPHYRA/lib/stm32lib/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_rtc.o build-OPHYRA/lib/stm32lib/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_rtc_ex.o build-OPHYRA/lib/stm32lib/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_spi.o build-OPHYRA/lib/stm32lib/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_tim.o build-OPHYRA/lib/stm32lib/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_tim_ex.o build-OPHYRA/lib/stm32lib/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_uart.o build-OPHYRA/lib/stm32lib/STM32F4xx_HAL_Driver/Src/stm32f4xx_ll_usb.o build-OPHYRA/lib/stm32lib/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_sd.o build-OPHYRA/lib/stm32lib/STM32F4xx_HAL_Driver/Src/stm32f4xx_ll_sdmmc.o build-OPHYRA/lib/stm32lib/STM32F4xx_HAL_Driver/Src/stm32f4xx_ll_fmc.o build-OPHYRA/lib/stm32lib/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_mmc.o build-OPHYRA/lib/stm32lib/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_sdram.o build-OPHYRA/ [... cut ...] e/examplemodule.o build-OPHYRA/lib/oofatfs/ff.o build-OPHYRA/lib/oofatfs/ffunicode.o build-OPHYRA/pins_OPHYRA.o -lstdc++ "c:/program files (x86)/gnu arm embedded toolchain/10 2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/thumb/v7e-m+fp/hard/libgcc.a"'
make: *** [Makefile:690: build-OPHYRA/firmware.elf] Error 1
```
I have investigated a little in the makefile, in the line that marks the error and I see that it is a problem with the **firmware.elf** generation.

Has anyone had a similar error?

stinos, 2021-03-02:
This part in the command:
```
-Lc:/ ./ (x86)/ ./ ./ toolchain/ 2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard/
```
is clearly wrong and could have something to do with spaces or escape characters in the path.
Can you provide more information about your build environment? What did you install, which commands you use to build, etc?

Krasn4ck, 2021-03-02:
> This part in the command:
>
> ```
> -Lc:/ ./ (x86)/ ./ ./ toolchain/ 2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard/
> ```
>
> is clearly wrong and could [...]

Ohh sure this is the command and my enviroment is cygwin in Win10
`make USER_C_MODULES=../../examples/usercmodule CFLAGS_EXTRA=-DMODULE_CEXAMPLE_ENABLED=1 all BOARD=OPHYRA -j8`

stinos, 2021-03-03:
The problem lies rather in the toolchain; how did you install that? How doe you let Cygwin know where to find the toolchain etc?

Krasn4ck, 2021-03-03:
The toolchain is in the win 10 path, but if I do a clean compile without adding external modules in C everything works normal and generates the .dfu and .hex files without any problem.
Look the clean compilation just the command `make BOARD=OPHYRA -j8` this is the result:
```
LINK build-OPHYRA/firmware.elf
   text    data     bss     dec     hex filename
 333428      16   27112  360556   5806c build-OPHYRA/firmware.elf
GEN build-OPHYRA/firmware0.bin
GEN build-OPHYRA/firmware1.bin
GEN build-OPHYRA/firmware.hex
GEN build-OPHYRA/firmware.dfu
```

stinos, 2021-03-03:
If you run `make BOARD=OPHYRA V=1`, what does the link command look like (the one you posted originally, which starts with arm-none-eabi-ld)?

Just an idea but if you remove the line `LDFLAGS_MOD += $(LDFLAGS_USERMOD)` from py/py.mk and try to build the modules does that change anything? Or if you try to build for another board?

Krasn4ck, 2021-03-04:
> If you run `make BOARD=OPHYRA V=1`, what does the link command look like (the one you posted originally, which starts with arm-none-eabi-ld)?
>
> Just an idea but if you remove the line `LDFLAGS_MOD += [...]
Sure is it:
```
LINK build-OPHYRA/firmware.elf
arm-none-eabi-ld -nostdlib -L boards -Tboards/stm32f407.ld -Tboards/common_ifs.ld -Map=build-OPHYRA/firmware.map --cref --defsym=_estack_reserve=8 --gc-sections -Lc:/ ./ (x86)/ ./ ./ toolchain/ 2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard/  -o build-OPHYRA/firmware.elf build-OPHYRA/py/mpstate.o build-OPHYRA/py/nlr.o build-OPHYRA/py/nlrx86.o build-OPHYRA/py/nlrx64.o build-OPHYRA/py/nlrthumb.o build-OPHYRA/py/nlrpowerpc.o build-OPHYRA/py/nlrxtensa.o build-OPHYRA/py/nlrsetjmp.o build-OPHYRA/py/malloc.o build-OPHYRA/py/gc.o build-OPHYRA/py/pystack.o build-OPHYRA/py/qstr.o build-OPHYRA/py/vstr.o build-OPHYRA/py/mpprint.o build-OPHYRA/py/unicode.o build-OPHYRA/py/mpz.o build-OPHYRA/py/reader.o build-OPHYRA/py/lexer.o build-OPHYRA/py/parse.o build-OPHYRA/py/scope.o build-OPHYRA/py/compile.o build-OPHYRA/py/emitcommon.o build-OPHYRA/py/emitbc.o build-OPHYRA/py/asmbase.o build-OPHYRA/py/asmx64.o build-OPHYRA/py/emitnx64.o build-OPHYRA/py/asmx86.o build-OPHYRA/py/emitnx86.o build-OPHYRA/py/asmthumb.o build-OPHYRA/py/emitnthumb.o build-OPHYRA/py/emitinlinethumb.o build-OPHYRA/py/asmarm.o build-OPHYRA/py/emitnarm.o build-OPHYRA/py/asmxtensa.o build-OPHYRA/py/emitnxtensa.o build-OPHYRA/py/emitinlinextensa.o build-OPHYRA/py/emitnxtensawin.o build-OPHYRA/py/formatfloat.o build-OPHYRA/py/parsenumbase.o build-OPHYRA/py/parsenum.o build-OPHYRA/py/emitglue.o build-OPHYRA/py/persistentcode.o build-OPHYRA/py/runtime.o build-OPHYRA/py/runtime_utils.o build-OPHYRA/py/scheduler.o build-OPHYRA/py/nativeglue.o build-OPHYRA/py/pairheap.o build-OPHYRA/py/ringbuf.o build-OPHYRA/py/stackctrl.o build-OPHYRA/py/argcheck.o build-OPHYRA/py/warning.o build-OPHYRA/py/profile.o build-OPHYRA/py/map.o build-OPHYRA/py/obj.o build-OPHYRA/py/objarray.o build-OPHYRA/py/objattrtuple.o build-OPHYRA/py/objbool.o build-OPHYRA/py/objboundmeth.o build-OPHYRA/py/objcell.o build-OPHYRA/py/objclosure.o build-OPHYRA/py/objcomplex.o build-OPHYRA/py/objdeque.o build-OPHYRA/py/objdict.o build-OPHYRA/py/objenumerate.o build-OPHYRA/py/objexcept.o build-OPHYRA/py/objfilter.o build-OPHYRA/py/objfloat.o build-OPHYRA/py/objfun.o build-OPHYRA/py/objgenerator.o build-OPHYRA/py/objgetitemiter.o build-OPHYRA/py/objint.o build-OPHYRA/py/objint_longlong.o build-OPHYRA/py/objint_mpz.o build-OPHYRA/py/objlist.o build-OPHYRA/py/objmap.o build-OPHYRA/py/objmodule.o build-OPHYRA/py/objobject.o build-OPHYRA/py/objpolyiter.o build-OPHYRA/py/objproperty.o build-OPHYRA/py/objnone.o build-OPHYRA/py/objnamedtuple.o build-OPHYRA/py/objrange.o build-OPHYRA/py/objreversed.o build-OPHYRA/py/objset.o build-OPHYRA/py/objsingleton.o build-OPHYRA/py/objslice.o build-OPHYRA/py/objstr.o build-OPHYRA/py/objstrunicode.o build-OPHYRA/py/objstringio.o build-OPHYRA/py/objtuple.o build-OPHYRA/py/objtype.o build-OPHYRA/py/objzip.o build-OPHYRA/py/opmethods.o build-OPHYRA/py/sequence.o build-OPHYRA/py/stream.o build-OPHYRA/py/binary.o build-OPHYRA/py/builtinimport.o build-OPHYRA/py/builtinevex.o build-OPHYRA/py/builtinhelp.o build-OPHYRA/py/modarray.o build-OPHYRA/py/modbuiltins.o build-OPHYRA/py/modcollections.o build-OPHYRA/py/modgc.o build-OPHYRA/py/modio.o build-OPHYRA/py/modmath.o build-OPHYRA/py/modcmath.o build-OPHYRA/py/modmicropython.o build-OPHYRA/py/modstruct.o build-OPHYRA/py/modsys.o build-OPHYRA/py/moduerrno.o build-OPHYRA/py/modthread.o build-OPHYRA/py/vm.o build-OPHYRA/py/bc.o build-OPHYRA/py/showbc.o build-OPHYRA/py/repl.o build-OPHYRA/py/smallint.o build-OPHYRA/py/frozenmod.o build-OPHYRA/extmod/moduasyncio.o build-OPHYRA/extmod/moductypes.o build-OPHYRA/extmod/modujson.o build-OPHYRA/extmod/modure.o build-OPHYRA/extmod/moduzlib.o build-OPHYRA/extmod/moduheapq.o build-OPHYRA/extmod/modutimeq.o build-OPHYRA/extmod/moduhashlib.o build-OPHYRA/extmod/moducryptolib.o build-OPHYRA/extmod/modubinascii.o build-OPHYRA/extmod/virtpin.o build-OPHYRA/extmod/machine_mem.o build-OPHYRA/extmod/machine_pinbase.o build-OPHYRA/extmod/machine_signal.o build-OPHYRA/extmod/machine_pulse.o build-OPHYRA/extmod/machine_i2c.o build-OPHYRA/extmod/machine_spi.o build-OPHYRA/extmod/modbluetooth.o build-OPHYRA/extmod/modussl_axtls.o build-OPHYRA/extmod/modussl_mbedtls.o build-OPHYRA/extmod/modurandom.o build-OPHYRA/extmod/moduselect.o build-OPHYRA/extmod/moduwebsocket.o build-OPHYRA/extmod/modwebrepl.o build-OPHYRA/extmod/modframebuf.o build-OPHYRA/extmod/vfs.o build-OPHYRA/extmod/vfs_blockdev.o build-OPHYRA/extmod/vfs_reader.o build-OPHYRA/extmod/vfs_posix.o build-OPHYRA/extmod/vfs_posix_file.o build-OPHYRA/extmod/vfs_fat.o build-OPHYRA/extmod/vfs_fat_diskio.o build-OPHYRA/extmod/vfs_fat_file.o build-OPHYRA/extmod/vfs_lfs.o build-OPHYRA/extmod/utime_mphal.o build-OPHYRA/extmod/uos_dupterm.o build-OPHYRA/lib/embed/abort_.o build-OPHYRA/lib/utils/printf.o build-OPHYRA/build-OPHYRA/frozen_content.o build-OPHYRA/lib/libc/string0.o build-OPHYRA/lib/mp-readline/readline.o build-OPHYRA/lib/netutils/netutils.o build-OPHYRA/lib/netutils/trace.o build-OPHYRA/lib/netutils/dhcpserver.o build-OPHYRA/lib/timeutils/timeutils.o build-OPHYRA/lib/utils/gchelper_native.o build-OPHYRA/lib/utils/pyexec.o build-OPHYRA/lib/utils/interrupt_char.o build-OPHYRA/lib/utils/sys_stdio_mphal.o build-OPHYRA/lib/utils/mpirq.o bu [... cut ...] ple/examplemodule.o build-PYBV10/cppexample/examplemodule.o build-PYBV10/lib/oofatfs/ff.o build-PYBV10/lib/oofatfs/ffunicode.o build-PYBV10/lib/littlefs/lfs2.o build-PYBV10/lib/littlefs/lfs2_util.o build-PYBV10/pins_PYBV10.o  "c:/program files (x86)/gnu arm embedded toolchain/10 2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/thumb/v7e-m+fp/hard/libgcc.a"'
make: *** [Makefile:690: build-PYBV10/firmware.elf] Error 1

```
I tried comment `LDFLAGS_MOD += $(LDFLAGS_USERMOD)` but is the same error.

stinos, 2021-03-04:
Sorry I meant the link command when not building user C modules. Anyway, I tried it and can reproduce it. The problem is that the C++ example modules is also included and casues the incorrect linker flags somehow. Not sure how yet, but one workaround is to delete usercmodule/cppexample.

stinos, 2021-03-04:
This patch works for me:
```
--- a/ports/stm32/Makefile
+++ b/ports/stm32/Makefile
@@ -145,7 +145,8 @@ endif
 CXXFLAGS += $(filter-out -Wmissing-prototypes -Wold-style-definition -std=gnu99,$(CFLAGS))
 CXXFLAGS += $(CXXFLAGS_MOD)
 ifneq ($(SRC_CXX)$(SRC_MOD_CXX),)
-LDFLAGS += -L$(dir $(shell $(CXX) $(CXXFLAGS) -print-file-name=libstdc++.a))
+LIBSTDCPP_FILE_NAME = "$(shell $(CXX) $(CXXFLAGS) -print-file-name=libstdc++.a)"
+LDFLAGS += -L"$(shell dirname $(LIBSTDCPP_FILE_NAME))"
 endif

 # Options for mpy-cross
```

I tried various things, but the thing is `make`'s `dir` command is built to take spaces as separator bewteen its arguments so nothing works out. `dirname` is POSIX so I guess this is an acceptable fix.

On the other hand, there are other Makefiles with the same or similar problems, and paths with spaces are just quite the mess when it comes to `make`; by far the easiest is to not use them.

@dpgeorge would you prefer a patch or rather a documentation update saying to not uses spaces?

Krasn4ck, 2021-03-04:
> +LIBSTDCPP_FILE_NAME = "$(shell $(CXX) $(CXXFLAGS) -print-file-name=libstdc++.a)"
> +LDFLAGS += -L"$(shell dirname $(LIBSTDCPP_FILE_NAME))"

Added this lines in the makefile of `ports/stm32` and really works.

```
CXX ../../examples/usercmodule/cppexample/example.cpp
CC ../../lib/stm32lib/CMSIS/STM32F4xx/Source/Templates/system_stm32f4xx.c
CC system_stm32.c
AS resethandler.s
AS ../../lib/utils/gchelper_m3.s
CC ../../examples/usercmodule/cexample/examplemodule.c
CC ../../examples/usercmodule/cppexample/examplemodule.c
CC ../../lib/oofatfs/ff.c
CC ../../lib/oofatfs/ffunicode.c
CC build-OPHYRA/pins_OPHYRA.c
CC build-OPHYRA/frozen_content.c
LINK build-OPHYRA/firmware.elf
   text    data     bss     dec     hex filename
 327924      16   27244  355184   56b70 build-OPHYRA/firmware.elf
GEN build-OPHYRA/firmware0.bin
GEN build-OPHYRA/firmware1.bin
GEN build-OPHYRA/firmware.hex
GEN build-OPHYRA/firmware.dfu

```
The modules in C was added the compilation.

dpgeorge, 2021-03-11:
> would you prefer a patch or rather a documentation update saying to not uses spaces?

That patch looks OK to me, I can't think of anything better.  It follows what you did in da71f55e23a97102d241481e410fdb22d7c55758 .  IMO it's fair enough to provide a "best effort" when it comes to spaces in paths.

closed 2021-03-25 by Krasn4ck