micropython/micropython #6980: Pyboard.py doesn't keep file path with folders on windows

martinlombana, 2021-03-02:
Hi, I am using the latest pyboard.py and, per the documentation, it will keep the path, providing that the folder exists also on destination, but it does not.

When executing this:

```
python pyboard.py --device COM4 -f cp srptools/context.py :
```
We get this, and the file is copied to the root.

```
cp srptools/context.py :context.py
```

Workaround: Specify the destination path manually. 

```
python pyboard.py --device COM4 -f cp srptools/context.py :srptools/context.py
```

As a bonus, it would be nice if pyboard.py created the destination path if it doesn't exist. 

Thanks!

dpgeorge, 2021-03-11:
> per the documentation, it will keep the path, providing that the folder exists also on destination

Can you show which part of the docs state this?

The standard behaviour of unix `cp` is to not keep the path, and pyboard.py follows that behaviour.

martinlombana, 2021-03-11:
[image]

https://docs.micropython.org/en/latest/reference/pyboard.py.html

In this example, it shows that. Unless I am reading something wrong. I had the folder created in the destination (mcu) so I expected it to work, as the example.

dpgeorge, 2021-03-11:
Ah, I see, thanks for that screenshot.  It is wrong (`lib/foo.py` will copy just `foo.py` to the root of the device).

@jimmo was that a typo?

martinlombana, 2021-03-11:
Ans is there a way to copy something to a folder or a folder completely?

Thanks!
-- 

___________________________________

*Martín Lombana*

*TODOLOGIC.COM <http://TODOLOGIC.COM> | APPS & MORE*

*+34 633 63 02 35  |   *+34 946 759 137

*INSTAGRAM *→* @ <https://instagram.com/todologic>TODOLOGIC
<https://instagram.com/todologic>*

dpgeorge, 2021-03-12:
> is there a way to copy something to a folder

You can do:
```
$ pyboard.py -f cp foo.py :lib/
```

martinlombana, 2021-03-12:
And all the contents of a folder? to a destination folder?
Thanks,