micropython/micropython #6981: ESP32: Expose ext1 wakeup status

excitedleigh, 2021-03-02:
Hello! I'm trying to build a battery-powered button panel that has four buttons on it, and comes out of sleep and does a bunch of network IO when one button is pressed.

At the moment I'm using `esp32.wake_on_ext1` to get the machine to wake when one of the buttons is pressed, and then I'm just reading from the pins the buttons are wired to. Unfortunately a brief press on the button seems to take less time than MicroPython does to wake, so the pin is no longer high at that point.

The ESP-IDF has a function [`esp_sleep_get_ext1_wakeup_status`](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/sleep_modes.html#_CPPv432esp_sleep_get_ext1_wakeup_status) which returns a bit mask of all the pins that triggered an EXT1 wake—can this be added to the MicroPython `esp32` module?

brendenriggs, 2021-06-23:
Seconded.  This is the only thing holding me back from using MP for my company's next project.

hueyy, 2022-11-22:
See https://github.com/micropython/micropython/pull/7990

ned-pcs, 2024-07-10:
At least on the ESP32-S3, you could just read the RTC wakeup register at 0x600080e4:
```py
def ext1_wakeup_pins():
    """return bitmask of GPIO0..GPIO21"""
    return mem32[0x600080e4]
```
The result is a bitmap, with the LS bit representing `RTC_GPIO0` (which is the same as `GPIO0`)