micropython/micropython #6983: UNIX: Running MPY files - Invalid Syntax

ALEX010001, 2021-03-03:
Hi, 
im  trying to run .mpy files - that generated as a result of running `mpy-cross myscript.py` .

I want to run this files directrly from micropython, i.e, 
I need to send to my microcontroller only a `.mpy` files and to run them directly.

how can i do it?

thanks.

(i didnt find an option to ask questions on the forum, so im asking it here)

bertelschmitt, 2021-03-04:
Did you run  mpy_cross with the -march=[your platform] option?

ALEX010001, 2021-03-04:
Yes i got `.mpy` file, and I want to running it directly.
I want something like:

./micropython myscript.mpy

how can i do that?

ALEX010001, 2021-03-04:
I need i due to lack of memory in my mcu.
I want to flash only the part that runs the bytecodes (.mpy files), and to send to my mcu only this .mpy files.

dpgeorge, 2021-03-11:
If you want to run a .mpy locally using the unix port of MicroPython then you need to import it rather than run it directly.  Eg:
```
$ mpy-cross -mcache-lookup-bc test.py
$ micropython -m test
```

jonnor, 2024-08-18:
I believe the above documents how to resolve the problem. Proposing to close this issue.

closed 2024-08-18 by robert-hh