micropython/micropython #6984: No Connection to Wifi with AUTH_WPA_WPA2_PSK

Killerbeast01, 2021-03-03:
Hey, 
first of all: i really love your work! 
Every time i work on esp´s i complain about the c-language, but with micropython i can programm the devices in my favourite lanuage :)

Since iam using this firmware, i got no issues, but tody at work i faced the first:

the esp8266 with the latest stable version (1.14) can't connect to a Wifi, secured with the authmode "AUTH_WPA_WPA2_PSK".
But if i setup a wifi with the same credentials, but with "AUTH_WPA2_PSK", the device connects nearly instant...

Is there something i have to consider in my code? 
Thank you :)

PS.: Sorry for my bad english ^^

jonnor, 2024-09-13:
Hi and apologies for no answers. Did you manage to resolve your question?

Killerbeast01, 2024-09-13:
Hi, iam sorry, we ended up with going back to c for this project... idk if the pullrequest here fixed it, and iam not in the environment anymore to check it :/ sry