micropython/micropython #6987: Board definitions for rp2 ports

tannewt, 2021-03-04:
Hi! Is there a plan for supporting other RP2040 boards besides the Pico? I don't see a `boards` directory in `ports/rp2`.

We're releasing the Feather RP2040 today and would like to have MicroPython support as well as CircuitPython. I suspect other vendors would like this as well.

I tried putting the Pico UF2 of MicroPython on the Feather but didn't get a serial port to appear. I suspect the challenge of supporting other boards will be the varying flash chips that different boards use. (The Feather uses a GD2564C.)

For CircuitPython, I've just started a TOML "database" for settings of flash chips [here](https://github.com/adafruit/nvm.toml).

DavidMcColl278, 2021-03-22:
Is there a temporary work around by adjusting make flags as was done to get C code to boot properly on the Feather RP2040?

see:
    Re: Feather RP2040 won't run with C code
    So it looks like just running: cmake .. -DPICO_BOARD=adafruit_feather_rp2040 -DPICO_DEFAULT_BOOT_STAGE2_FILE=<path-to-sdk>/pico-sdk/src/rp2_common/boot_stage2/boot2_generic_03h.S will do the trick.

tannewt, 2021-03-22:
The flash difference is usually the quad enable bit so you can use CircuitPython once to set it and then load other code after.

dpgeorge, 2021-04-09:
See #7099 for a PR for this feature.  It looks like the feather rp2040 is supported.

dpgeorge, 2021-04-13:
Implemented in 7ca686684ea5bffb45764bc3782f54133855a3c9

closed 2021-04-13 by dpgeorge