micropython/micropython #6988: esp32: gc.collect() causes panic

bertelschmitt, 2021-03-04:
On a bog-standard DOIT ESP32 Devkit V1 (which is nothing else than a vanilla ESP32-WROOM-32 on a board, no extra RAM or memory) gc.collect() makes the part to crash when flashed with one of the more recent MicropPython images. This is with today's esp32-20210304-unstable-v1.14-83-g680ce4532.bin

```
>>> import gc
>>> gc.collect()
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40156436  PS      : 0x00060930  A0      : 0x80154620  A1      : 0x3ffccd40  
A2      : 0x0f32f602  A3      : 0x000000c1  A4      : 0x00000030  A5      : 0x00000001  
A6      : 0x00000018  A7      : 0x00000000  A8      : 0x3ffccee0  A9      : 0x3ffcce60  
A10     : 0x3f403cc8  A11     : 0x3ffe59c0  A12     : 0x3ffccee0  A13     : 0x3f403cd0  
A14     : 0x3ffcd060  A15     : 0x3ffcd080  SAR     : 0x00000002  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0f32f606  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff  

Backtrace:0x40156433:0x3ffccd40 0x4015461d:0x3ffccd70 0x4015488d:0x3ffccdb0 0x401c1a0a:0x3ffccdd0 0x400db52e:0x3ffccdf0 0x400e17ad:0x3ffcce10 0x400e1895:0x3ffcce30 0x400e24a4:0x3ffcce60 0x400d7d54:0x3ffccee0 0x400f3dc4:0x3ffccf20 0x400ff13b:0x3ffccf40 0x400db4f5:0x3ffccf60 0x400e17ad:0x3ffccf80 0x400e18da:0x3ffccfa0 0x400e3e75:0x3ffccfc0 0x400db684:0x3ffcd060 0x400e17ad:0x3ffcd0b0 0x400e17d6:0x3ffcd0d0 0x400f3807:0x3ffcd0f0 0x400f3b15:0x3ffcd180 0x400d5724:0x3ffcd1c0

ELF file SHA256: 2fc7913b0d94aa3c

Rebooting...
```

No such problem with higher-RAM spiram parts.

robert-hh, 2021-03-04:
Are you using the GENERIC or GENERIC-SPIRAM images for your test on the WROOM part.

bertelschmitt, 2021-03-04:
Generic. Generic spiram would  be : esp32spiram-2021030 ....

bertelschmitt, 2021-03-05:
All esp32 GENERIC firmwares  since and including  esp32-20210222-unstable-v1.14-80-g75db0b907.bin have that problem. Last good was  esp32-idf4-20210206-unstable-v1.14-9-g9dedcf122.bin
I have a whole drawer of these parts. They all show the same problem.

robert-hh, 2021-03-05:
No problem here with the esp32-20210302.. and esp32-20210305.. version on various wemos boards.#
I just try:
import gc
gc.collect()

robert-hh, 2021-03-05:
When I try to decode the backtrace, I get the below. It does not look reasonable. But that may be caused by the fact, that I do not have the original elf file, but only one which I created myself.
```
reading backtrace from stdin
0x40156433:0x3ffccd40 0x4015461d:0x3ffccd70 0x4015488d:0x3ffccdb0 0x401c1a0a:0x3ffccdd0 0x400db52e:0x3ffccdf0 0x400e17ad:0x3ffcce10 0x400e1895:0x3ffcce30 0x400e24a4:0x3ffcce60 0x400d7d54:0x3ffccee0 0x400f3dc4:0x3ffccf20 0x400ff13b:0x3ffccf40 0x400db4f5:0x3ffccf60 0x400e17ad:0x3ffccf80 0x400e18da:0x3ffccfa0 0x400e3e75:0x3ffccfc0 0x400db684:0x3ffcd060 0x400e17ad:0x3ffcd0b0 0x400e17d6:0x3ffcd0d0 0x400f3807:0x3ffcd0f0 0x400f3b15:0x3ffcd180 0x400d5724:0x3ffcd1c0

BT-0: 0x40156433 is in ipcp_addci (/home/robert/Downloads/esp-idf/components/lwip/lwip/src/netif/ppp/ipcp.c:895).
895	    ADDCIADDR(CI_ADDR, go->neg_addr, go->ouraddr);
BT-1: 0x4015461d is in lwip_netconn_do_close_internal (/home/robert/Downloads/esp-idf/components/lwip/lwip/src/api/api_msg.c:1096).
1096	    conn->current_msg->err = err;
BT-2: 0x4015488d is in netconn_drain (/home/robert/Downloads/esp-idf/components/lwip/lwip/src/api/api_msg.c:859).
859	{
BT-3: 0x401c1a0a is in __ieee754_sinhf (/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libm/math/ef_sinh.c:56).
56	/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libm/math/ef_sinh.c: Datei oder Verzeichnis nicht gefunden.
BT-4: 0x400db52e is in fun_bc_call (/home/robert/Downloads/MicroPython/micropython/py/objfun.c:260).
260	    DECODE_CODESTATE_SIZE(self->bytecode, n_state, state_size);
BT-5: 0x400e17ad is in mp_load_method_maybe (/home/robert/Downloads/MicroPython/micropython/py/runtime.c:1086).
1086	    if (attr == MP_QSTR___class__) {
BT-6: 0x400e1895 is in mp_import_all (/home/robert/Downloads/MicroPython/micropython/py/runtime.c:1449).
1449	    for (size_t i = 0; i < map->alloc; i++) {
BT-7: 0x400e24a4 is in m_malloc_fail (/home/robert/Downloads/MicroPython/micropython/py/runtime.c:1507).
1507	    if (gc_is_locked()) {
BT-8: 0x400d7d54 is in gc_free (/home/robert/Downloads/MicroPython/micropython/py/gc.c:594).
594	        FTB_CLEAR(block);
BT-9: 0x400f3dc4 is in f_read (/home/robert/Downloads/MicroPython/micropython/lib/oofatfs/ff.c:3629).
3629	    if (btr > remain) btr = (UINT)remain;       /* Truncate btr by remaining bytes */
BT-10: 0x400ff13b is in emit_native_global_exc_entry (/home/robert/Downloads/MicroPython/micropython/py/emitnative.c:1240).
1240	            ASM_MOV_REG_LOCAL_ADDR(emit->as, REG_ARG_1, 2);
BT-11: 0x400db4f5 is in fun_bc_call (/home/robert/Downloads/MicroPython/micropython/py/objfun.c:248).
248	STATIC mp_obj_t fun_bc_call(mp_obj_t self_in, size_t n_args, size_t n_kw, const mp_obj_t *args) {
BT-12: 0x400e17ad is in mp_load_method_maybe (/home/robert/Downloads/MicroPython/micropython/py/runtime.c:1086).
1086	    if (attr == MP_QSTR___class__) {
BT-14: 0x400e3e75 is in mp_execute_bytecode (/home/robert/Downloads/MicroPython/micropython/py/vm.c:851).
851	                    sp -= unum - 1;
BT-15: 0x400db684 is in mp_obj_new_fun_bc (/home/robert/Downloads/MicroPython/micropython/py/objfun.c:400).
400	        o->extra_args[n_def_args] = def_kw_args;
BT-16: 0x400e17ad is in mp_load_method_maybe (/home/robert/Downloads/MicroPython/micropython/py/runtime.c:1086).
1086	    if (attr == MP_QSTR___class__) {
BT-17: 0x400e17d6 is in mp_load_method_maybe (/home/robert/Downloads/MicroPython/micropython/py/runtime.c:1099).
1099	        type->attr(obj, attr, dest);
BT-18: 0x400f3807 is in cmp_lfn (/home/robert/Downloads/MicroPython/micropython/lib/oofatfs/ff.c:1866).
1866	    if ((dir[LDIR_Ord] & LLEF) && wc && lfnbuf[i]) return 0;    /* Last segment matched but different length */
BT-19: 0x400f3b15 is in dir_register (/home/robert/Downloads/MicroPython/micropython/lib/oofatfs/ff.c:2523).
2523	    nent = (sn[NSFLAG] & NS_LFN) ? (nlen + 12) / 13 + 1 : 1;    /* Number of entries to allocate */
BT-20: 0x400d5724 is in machine_pin_obj_init_helper (/home/robert/Downloads/MicroPython/micropython/ports/esp32/machine_pin.c:188).
188	            gpio_pullup_en(self->id);
```

bertelschmitt, 2021-03-06:
@robert-hh:
Sorry, I could have sworn that  I erased before flashing, but for good measure, I erased again, flashed, and:

```
MPY: soft reboot
MicroPython v1.14 on 2021-03-04; ESP32 module with ESP32
Type "help()" for more information.
>>> import gc
>>> gc.collect()
>>>
```

My apologies. Close issue.

closed 2021-03-11 by dpgeorge