micropython/micropython #6997: Esp8266 doesn't execute code after woke up from deep sleep

lazarvgd, 2021-03-05:
I have a problem with waking ESP8266 from deep sleep. Here is my code that i run:
`
import network
import machine
import time
from machine import RTC
import ntptime
from wdt import WDT

def deep_sleep(msecs):
  #configure RTC.ALARM0 to be able to wake the device
  rtc = machine.RTC()
  rtc.irq(trigger=rtc.ALARM0, wake=machine.)
  # set RTC.ALARM0 to fire after Xmilliseconds, waking the device
  rtc.alarm(rtc.ALARM0, msecs)
  #put the device to sleep
  machine.deepsleep()

wdt=WDT()
led = machine.Pin(2, machine.Pin.OUT)

sta_if = network.WLAN(network.STA_IF); sta_if.active(True)
sta_if.scan()                             # Scan for available access points
sta_if.connect("TS-C99W", "VyHPS8ZJ") # Connect to an AP
sta_if.isconnected()    
print(sta_if.isconnected())
rtc = RTC()

while True:
    if machine.reset_cause() == machine.DEEPSLEEP_RESET:
        print('woke from a deep sleep')
    else:
        print('power on or hard reset')
    for i in range(10):
        led.off()
        time.sleep(1)
        led.on()
        time.sleep(1)
    print(sta_if.isconnected())
    ntptime.settime() # set the rtc datetime from the remote server
    print(rtc.datetime())    # get the date and time in UTC
    time.sleep(0.25)
    deep_sleep(20000)
    wdt.feed()
`

The output that I am getting is this:

[image]

The board that I am using is this:
  
[image]

I am not sure why is this happening? What am I doing wrong?
Please help me to solve this issue.

Thanks

dpgeorge, 2021-03-11:
It's best to ask these kinds of questions at https://forum.micropython.org

Did you connect GPIO16 to the RESET pin?  That is required.

Also, might be simpler to use `machine.deepsleep(msecs)` which automatically configures the alarm.

kevinkm, 2021-06-02:
You have to connect GPIO16 to RST with a 10k resistor, otherwise it will consume more power while it starts to work as you may know.