micropython/micropython #6998: rp2: uart.read() never returns

kevinkk525, 2021-03-05:
In the current implementation calling uart.read() without an expected amount of bytes will never return.
Expected behaviour: As soon as some data was read, that data should be returned.

In https://github.com/micropython/micropython/pull/6870 @robert-hh states:

> When uart.read() is called without an argument, the size at the machine_uart_read() is 256.

However, I can't reproduce that. Sending over 400 Bytes in small batches or a big one still doesn't return anything.

peterhinch, 2021-03-06:
Anyone following this up might want to look at [this forum thread](https://forum.micropython.org/viewtopic.php?f=21&t=9977) which has more detail on test conditions.

kevinkk525, 2021-03-06:
I opened it more for completeness sake so nobody else might stumble on the problem. @robert-hh fixed it in his PR #6870 even though the title doesn't mention it.

dpgeorge, 2021-03-11:
This should be fixed now by the addition of timeout support to UART.  See 8ade163fffe9cca620ebe945dec8d25c54dc877d

Feel free to reopen if behaviour is not as expected.

closed 2021-03-11 by dpgeorge