micropython/micropython #6999: rp2: uart performance problem with higher baudrates or when doing additional operations

kevinkk525, 2021-03-05:
The rp2 suffers from a performance problem reading from the UART, most likely getting received bytes out of the FIFO and into the RAM in time. (At least that's what I think because I doubt that between an esp32 and rp2 the uart can get out of sync with only 120 Bytes transmitted and even if it did, I should read garbage and not nothing at all. So my assumption is dropped bytes)
The problem becomes more obvious the higher the baudrate is.

Most obvious test case at baudrate 460800:
```
# on the rp2:
uart = machine.UART(1, rx=machine.Pin(5), tx=machine.Pin(4), baudrate=460800)
def read(l):
    # function just to print the length of the received message and flush the uart before waiting for new bytes.
    while uart.any():
        uart.read(1)
    r=uart.read(l)
    print(r,len(r))
read(60)
b'012345678901234567890123456789012345678901234567890167890123' 60
```
On the esp32 (done manually on the repl):
```
>>> a=b'0123456789'
>>> uart = machine.UART(1, tx=17, rx=16, baudrate=460800)
>>> uart.write(a*6) # send 60 Bytes
>>> uart.write(a*6)
```
On the esp32 I had to send 60 Bytes twice (!) because the rp2 didn't receive all 60 Bytes during the first transmission. It seems to have dropped Bytes after 51 bytes because that's where the sequence starts again every time, so it's a consitent error.

Running the same test at baudrate 115200 was fine even up to 4000 Bytes. Therefore my assumption that this is a performance problem because if it was just getting out of sync, it wouldn't be fine with 4000 Bytes at baudrate 115200 but only less than 60 Bytes at 460800.

To further show the performance problem when doing some small operations while receiving data like reading in smaller chunks and merging the received strings:

Testcase (note: usually I'd use uart.read() but that doesn't work on the rp2 at the moment, so we're reading in chunks of 20):
```
def test2(t,t2):
...     while True:
...         gc.collect()
...         recv=b''
...         read=[]
...         st=time.ticks_us()
...         while time.ticks_diff(time.ticks_us(),st)<t:
...             if uart.any():
...                 rr=uart.read(20)
...                 recv+=rr
...                 read.append(len(rr))
...             else:
...                 time.sleep_us(t2)
...         if len(recv)>0:
...             print(recv,len(recv),read)
...
...
...
>>> test2(10000,0)
```
The output of always sending 60 Bytes manually after reading the last output results in an output of something like this on the rp2:
```
b'012345678901234567890123456789012345678901234567890123456789' 60 [20, 20, 20]
b'012345678901234567890123456789012345678901234567890123456789' 60 [20, 20, 20]
b'01234567890123456789' 20 [20]
b'0123456789012345678901234567890101234567' 40 [20, 20]
b'8901234567890123456789012345678901234567' 40 [20, 20]
b'8901234567890123456789012345678901234567' 40 [20, 20]
b'89012345678901234567' 20 [20]
```
As you can see, at first receiving 60 Bytes in chunks of 20 Bytes was working but only after a few more transmissions it screwed up and lost some bytes.
And that's at baudrate 115200! 

Repeating this test at baudrate 460800 already screws up when sending 40 Bytes at once! The output is then:
```
b'0123456789012345678901234567890101234567' 40 [20, 20]
```

So suddenly even the previously stable uart.read(4000) at baudrate 115200 is now unreliable when trying to read 60 Bytes in 20 Bytes chunks!

I'll spare you the comparison between two esp32s. The short version is: It's a lot better, I can send 800Bytes at baudrate 460800 without any issues.

So in the current uart implementation there seems to be a serious performance issue resulting in the uart dropping bytes, making it rather impossible to use as the smallest amount of operations already impacts its performance severly even at baudrate 115200.
I understand the buffer is only 32 Bytes but to fill that buffer the uart at baudrate 115200 needs ~2.5ms, which should be more than enough (after removing 20 Bytes from the buffer when reading the next chunk) to perform the operations in test2() since that's just a small merge of 2 byte objects with less than 40 Bytes lenght in total.

robert-hh, 2021-03-06:
So I did a little bit of timing analysis. The test set-up is a ESP32, sending 10 or more bytes at 460800 baud to a RP2. The send is manually started. The RP2 had a little program consisting in the main loop:
```
while True:
    v=u.read()
    if v is not None:
        print(v)
```
The timeout was set to 10 ms. In order to analyze the timing of the uart.read() call, I added some signalling by switching GPIO pins.
- on entry and leaving the function machine_uart_read() GPIO10 was set high and the low again
- around getting data from the UART GPIO11 was set high and low again.
- around the MICROPY_EVENT_POLL_HOOK macro GPIO8 was set high and low again

Findings:
- there is no buffering in the SDK layer. All buffering happens in the UART FIFO, which is 32 bytes long. That is the longest message that will be stored when no read is active.
- getting a byte from UART is fast. Actually is is just a register access and takes less than 100ns.
- the MICROPY_EVENT_POLL_HOOK macro is slow. It takes about 1 ms, meaning that most of the time of the machine_uart_read() is spent in the MICROPY_EVENT_POLL_HOOK macro, being blind for uarft data at this time. Together with the 32 byte FIFO that means, that the fastest baud rate the device can handle without data loss is about 32000, while being in the read call.
- trying the event hook mechanism used in the  PIO get function changes the picture. This hook takes about 2 µs. With that UART can easily cope with MBit baud rates. I do not know which downside it has using that poll hook. Something else has to wait then.

Attachments: 
- Basic timing showing the 10 ms timeout and the poll hooks
- Split timing where a pool hook ends in the middle of a transmission, such that some of the data is collected, but the rest after the next poll hook
- Timing using the fast poll hook, showing that UART data is collected as soon as it arrives

Side note: In the above test program, the time between calling uart.read() at python level varies between 42 an 700 µs.

[image]
[image]
[image]
[image]

kevinkk525, 2021-03-06:
Awesome analysis! Thanks a lot!

> trying the event hook mechanism used in the PIO get function changes the picture. This hook takes about 2 µs. With that UART can easily cope with MBit baud rates. I do not know which downside it has using that poll hook. Something else has to wait then.

Looking forward to a possibly faster uart implementation. Could you share your modifications? Or is it too early/too complex or too unclear which possible problems it could create?

robert-hh, 2021-03-06:
Of course. Replace in machine_uart.c the line 240:

            MICROPY_EVENT_POLL_HOOK

by

            mp_handle_pending(true);
            MICROPY_HW_USBDEV_TASK_HOOK

That could affect performance of other tasks only when machine_uart_read() is called.

kevinkk525, 2021-03-06:
Thanks. 
Strangely though it does not have any effect on my builds.. Not sure what I did wrong but I replaced it in your PR at line 240 in the read method. So I think I did it correctly. Even tried it with a build without your timeout PR and replaced the EVENT_POLL_HOOK in the read() method but same result..
Sending 60 Bytes at 480600 baudrate often results in only 32 Bytes being read, even when using read() without specifying an amount. Sending 40 Bytes works most of the time, sending 60 Bytes results in only 32 Bytes being read most of the time..

kevinkk525, 2021-03-06:
Thanks for posting the changes. Forget my last message, I shouldn't work when my head hurts too much :D I pasted the changes with wrong indentation so couldn't work correctly..
Now it works great! Can even send 800 Bytes in a single message.

robert-hh, 2021-03-06:
This faster polling is only effective if read() is actively waiting for data when it arrives. So you ahve call read() with a long timeout and then send the data. I agree that this is not always practical. So the only cure would be, if the firmware is able to collect a larger amount of data in the background. On ESP32 this is a interrupt service of RTOS. On rp2, this is not available. So it wold have to be implemented. The other option would be DMA transfer. That happens immediately. Using that you would set up a sufficiently large receive buffer, start DMA and check the number of received bytes. No call to uart.read() required.
At the moment I am preparing DMA examples for PIO. But tomorrow I should be able to make one for the UART. That should be more simple.

kevinkk525, 2021-03-06:
Yeah that is the biggest drawback at the moment. If you happen to do other operations while a transmission is incoming, you quickly lose Bytes with this small buffer. A DMA solution would be great. Thanks for your efforts.

robert-hh, 2021-03-07:
@kevinkk525 I added a DMA helper function for UART to rp2_util.py in https://github.com/robert-hh/RP2040-Examples/tree/master/rp2_util and have also an example how to use it with UART in the corresponding README.

kevinkk525, 2021-03-07:
@robert-hh thank you! This works great!
Makes a quite easy to use high level workaround for a reliable uart.

Would it be complicated to implement it in machine_uart.c too?

robert-hh, 2021-03-07:
> Would it be complicated to implement it in machine_uart.c too?

Not complicated, but it does not fit into the context of a that class. It would in a dedicated DMA module. But that does not exist in any port. A useful extension of the UART class would be a irq filled larger ring buffer, which extends the existing uart FIFO. That one would not affect the API and could be used to make the API more compatible to the other ports. Even things like rx_buf=nnn or the call to any() can then be set up in a compatible way.

kevinkk525, 2021-03-07:
I see.
Yeah my thought was to have a more compatible uart supporting rx_buf=nnn so there's less of a difference between the different micropython ports. If the DMA approach doesn't fit into the context, then I guess only the irq approach is left.

robert-hh, 2021-03-08:
I have made a first version of a machine_uart.c module, which uses irq for collection incoming data and stores it in a buffer. The default size is 256 bytes. The keyword rxbuf=nnnn on the uart instantiation is supported. uart.any will tell the actual numbers available for read. First tests seem fine. But I'm a little bit uncertain about irq handling with these two cores. Code beautifying may also be done.
This branch is based on the master branch, so timeout and invert are not included. I might have to merge this buffer branch manually into the timeout branch once. But for the moment I wanted to have it separated.
Whether irq for sendind data is needed, may be discussed. The advantage is it's non-blocking behavior.
https://github.com/robert-hh/micropython/tree/rp2_uart_buffer

kevinkk525, 2021-03-08:
Thanks! Looks good to me but I don't understand half of it anyway :D 
Will definitely try to test it the next days.

I was also thinking about possible implications of using two separate cores.. My goal was to run a communications library on the 2nd core so probably the irq and the library will end up running on different cores. But I guess it shouldn't be a problem for simple read/write as only single values are changed? (the ring buffer indices) Anyway, I don't know the micropython source code well enough to even get an idea of possible problems.

A non-blocking write method might be nice too, especially at lower baudrates but even at 115200 sending 256 Bytes (default txbuffer) would take around 20ms.

robert-hh, 2021-03-08:
The code contains pieces and ideas from the STM32 port. So I did not start from scratch. No magic!
I updated the file with some polishing and further checks. One other aspect I found was the collision with REPL, when enabled for UART. But in that case UART is anyhow not avalable for other  tasks.
Buffering transmit should be possible. But I'll do that later, when eventually the other PR are integrated.

davehylands, 2021-03-08:
As far as multiple cores go (having written lots of linux kernel drivers which run in SMP environments), generally you'd only run into problems when the getter and putter on the circular buffer are running on different cores, and even that can be alleviated by having a get pointer and put pointer that live on different cache lines, and not storing a length (i.e. try to design things so that you don't fields that are written by code on both cores at the same time).

If for some reason you do need to have a field be updated by both cores, then you'll need to introduce a spinlock. Disabling interrupts is generally not enough since it only disables interrupts on your own core.

kevinkk525, 2021-03-09:
Wouldn't in my scenario the getter and the putter run on different cores? The getter would be my read() on the 2nd core and the setter the irq on the first core, where I created the uart object. I don't understand all the code but it looks like setter and getter only update the pointers.

robert-hh, 2021-03-09:
As far as I understand the MP implementation for the Pico, the two cores are used for two threads. So if you start a second thread, it will run on the second core, and if you instatantiate the UART in the same thread which uses it, there is no problem with it. 
Almost all of today I tried to find the reason for a bug, which seemingly cause the uart IRQ to receive strange data. I had a test, sending exactly 1000 bytes with a known pattern (0123456789 100 times). The receiver would read 1000 bytes and check the pattern for correctness. It worked almost all of the time, but...  Sometimes part of the received data was shown as null bytes, even if the buffer would only contain ASCII, sometimes there were gaps. The latter could be explained with a race condition between the interrupt service routine, and the data collection by the program code. But not attempt to serialize worked. What worked was to add a gc.collect() in the main loop of the receiver. So there may be a problem in the memory management.
With that I was able to receive ~10000 blocks at 460800 baud up to now in a row. I will let it run through the night.

robert-hh, 2021-03-10:
I removed the branch rp2_uart_buffer and made a new branch rp2_uart, which includes support for timeout, invert and rx buffer. It also accepts the txbuf keyword in the instantiation and init() call, but it is silently ignored.
The buffered RX ran over night transferring 42GB in block of 1 k size each without any errors. That required frequent calls to gc.collect() in the test script.

peterhinch, 2021-03-10:
In terms of **uasyncio** stream I/O the need for buffering is only on the read side - write operations use a local buffer.

Read buffering is needed to spool data during the latency that occurs between the reading task yielding to the scheduler and it next gaining execution. The size of the buffer depends on the baudrate and the maximum latency, both of which are application dependent. Consequently the size should be user-specified.

kevinkk525, 2021-03-12:
Your PR #7022 looks great. Can send 2kB without problems with a few ms read delay at 460800. Thanks. Going to test it on the 2nd core soon (hopefully).
The memory problem requiring gc.collect() made me wonder: Can I even safely call gc.collect() from the 2nd core? What if it is called from both cores at the same time? Can't rely on user code to call it regularly enough to not cause problems..

robert-hh, 2021-03-12:
> The memory problem requiring gc.collect() made me wonder

These problems are gone. The file of the PR is stable in respect to that. I did not use it with two cores. But that should be safe if uart instatiation and use are on the same core. Maybe otherwise too.

kevinkk525, 2021-03-12:
That's great! Thanks. 
Yeah instantiating and running on the same core should work, then there's no cross-core code and the irq would be independent of any user-code running on the other core that could disable irqs.

kevinkk525, 2021-06-18:
So I guess this could be closed now. The new uart implementation works a lot better and problems with the implementation and thread were/are already discussed elsewhere.

closed 2021-06-18 by kevinkk525