micropython/micropython #7003: esp32: CMake v3.5 is not supported

amirgon, 2021-03-06:
According to esp-idf, [CMake version 3.5 or newer is required for use with ESP-IDF.](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-setup.html#codecell2:~:text=CMake%20version%203.5%20or%20newer%20is%20required%20for%20use%20with%20ESP%2DIDF.)

However, Micropython's CMake script uses syntax that is not available on CMake 3.5:

https://github.com/micropython/micropython/blob/680ce45323248df288ee8ebd055a4caacb3e46f3/ports/esp32/main/CMakeLists.txt#L112-L114

`GREATER_EQUAL` is [not available on CMake 3.5](https://cmake.org/cmake/help/v3.5/command/if.html) and was added only on 3.7.  
The error that CMake 3.5 displays in such case gives little clue about the source of the problem ("Unknown arguments specified").

It's best not to rely on CMake features that are not supported by CMake 3.5, or at least specify very clearly what is the minimal expected CMake version for building the ESP32 port.

dpgeorge, 2021-03-11:
Thanks for the report.

I guess there are 2 options: require a higher version of cmake (probably 3.8, enforced by `ports/esp32/CMakeLists.txt`), or change things so cmake 3.5 can be used.

Note that the use of `COMMAND_EXPAND_LISTS` is required by the zephyr port (see dff6fc64d23c548dce2b43096c3f1522db303de8) so an alternative would need to be found there.

From a maintenance point of view, the simplest thing would be to require a recent version of cmake.

Also would be good to enforce CI to use the specific version of cmake that is supported to test the build.

dpgeorge, 2021-04-22:
See #7145.

closed 2021-04-22 by amirgon