micropython/micropython #7006: rp2: build not including any modules from the USER_C_MODULES path

jsimonrichard, 2021-03-07:
I've been trying to build micropython using the rp2 port, and I would like to include ulab (https://github.com/v923z/micropython-ulab), a numpy-like array manipulation library. I've been able to build micropython without the library, but when I follow this documentation: https://docs.micropython.org/en/latest/develop/cmodules.html, the module never gets included into the build (micropython itself still works fine, though). I'm missing the line

```
Including User C Module from ../../path/to/module
```
in the output of my make command as stated in the aforementioned documentation.

Below is my file structure.
```
ulab/
    code/
        micropython.mk
        ulab.c
        
micropython/
    ports/
         rp2/
            makefiles, c code, etc.
```

And the make command I'm using (run in `micropython/ports/rp2`) is:
```
make USER_C_MODULES=../../../ulab/ all
```

**DISCLAIMER:** I'm a noob when it comes to micropython, so I apologize if I'm missing anything fundamental. This is my first project using micropython.

Also, I'm pretty sure this isn't a problem with ulab (although I could be wrong). There's a `micropython.mk` file and c code, so you would think the compiler would at least find the files and then throw an error. It's not throwing any errors, though.

imliubo, 2021-03-08:
Maybe this PR can help you. #6960

dpgeorge, 2021-04-13:
Support for building USER_C_MODULES on the rp2 port was added in 0cf12dd59c9ddddbd602d4267410033cb5a9d265

closed 2021-04-13 by dpgeorge