micropython/micropython #7007: esp32: The modified .py files can't be compile automatically.

lesson6, 2021-03-09:
when i change some python files,  and compile use command "make". Those files can't compile automatically.

lesson6, 2021-03-10:
hi，IhorNehrutsa. Thank you for answer.Yes,I mean the files in the dirctory micropython/ports/esp32/modules.
I used the latest micropython that suport cmake.
micropython commit id is  59a129f22f096d992496111c498b3ea97e637115.
 And esp-idf commit id is dddcc2ede8601915ad134ecf46f8682e5e6de2a4 (HEAD -> release/v4.2, origin/release/v4.2).
Maybe there have some different .

for example, I changed _boot.py(just add a line to print hello), then make, it show "no work to do".
[image]

lesson6, 2021-03-10:
By the way , it work well when i use makefile to compile earlier version.

dpgeorge, 2021-03-11:
Seems to be related to #6948.

lesson6, 2021-03-19:
Yes, It is the same reason.

closed 2021-03-19 by lesson6