micropython/micropython #7008: esp32: 'BLE' object has no attribute 'gap_pair'

z7jzk, 2021-03-10:
Generic ESP32 implementation. Tried on both v1.14 [esp32-idf4-20210202-v1.14.bin](https://micropython.org/resources/firmware/esp32-idf4-20210202-v1.14.bin) and v1.14-unstable latest [esp32-20210310-unstable-v1.14-84-g59a129f22.bin](https://micropython.org/resources/firmware/esp32-20210310-unstable-v1.14-84-g59a129f22.bin).

`self._ble.gap_pair(self._conn_handle)`

yields error:

`AttributeError: 'BLE' object has no attribute 'gap_pair'`

Is `gap_pair(conn_handle,/)` ready to use yet? I'm having issues in re-connecting to my bluetooth device as I need to pair it to add to trusted devices (connecting alone doesn't seem adequate). 

[ubluetooth — low-level Bluetooth: gap_pair](https://docs.micropython.org/en/latest/library/ubluetooth.html#ubluetooth.BLE.gap_pair)

dpgeorge, 2021-03-11:
Pairing/bonding is not yet supported on the esp32, because the esp32 needs a bit of work to switch it to synchronous BLE events.

z7jzk, 2021-03-12:
Thanks @dpgeorge! Any idea when it will be accessible on esp32? I'll see if an STM32 will work in the near term for our project.

dpgeorge, 2021-03-12:
> Any idea when it will be accessible on esp32?

No, sorry.  There are a lot of other things competing for development time!

z7jzk, 2021-03-15:
Understood. I was wondering directionally when this may be ready as it will affect the hardware we end up going with for our project.

dpgeorge, 2021-03-15:
I had a quick look at this the other day and made some progress, but it needs more time.  I might be able to make a patch for you to try out.

z7jzk, 2021-03-15:
That would be great! Keep me posted and I'll give it a shot.

dpgeorge, 2021-03-17:
Please see #7046 for work-in-progress on this feature.  @z7jzk any feedback on that would be great!

z7jzk, 2021-03-17:
I'll pull and start testing. I'll keep you posted on how it goes! Thank you for the quick response on this one!

z7jzk, 2021-03-18:
@dpgeorge, that did it! I was able to successfully pair to my peripheral bt device and the 2 devices connect on restart successfully each time, so it seems they are both added in the trusted devices. Need me to review your PR?

dpgeorge, 2021-03-18:
> and the 2 devices connect on restart successfully each time, so it seems they are both added in the trusted devices.

Did you implement the BLE secret IRQs to get this working?

dpgeorge, 2021-03-18:
> Need me to review your PR?

Not at this stage, thanks.  It's still a WIP.

z7jzk, 2021-03-19:
> Did you implement the BLE secret IRQs to get this working?

I did not, but it seems that the pairing functionality allows my peripheral device to store the esp32 I'm using as a central in the trusted devices, so perhaps that is why this is working now!

z7jzk, 2021-04-07:
> Did you implement the BLE secret IRQs to get this working?

@dpgeorge, I implemented the `_IRQ_GET_SECRET` and `_IRQ_SET_SECRET` events tonight to store the bonding secrets more permanently, so that I don't have to put my peripheral into pairing mode each time I connect (the device is able to connect without going into pairing mode, but the devices are not able to communicate with one another). 

I get the following error before the device reboots (this happens continuously):

```
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400da29d  PS      : 0x00060430  A0      : 0x800dc83b  A1      : 0x3ffcfd80
A2      : 0x3ffeb4a0  A3      : 0x3ffeb4a0  A4      : 0x00000000  A5      : 0x00000003
A6      : 0x3ffeb7cc  A7      : 0x00000064  A8      : 0x00000001  A9      : 0x3ffcfd60
A10     : 0x3ffcfee0  A11     : 0x3ffbd82c  A12     : 0x000000d7  A13     : 0x000004ea
A14     : 0x3ffcfd28  A15     : 0x00000002  SAR     : 0x00000004  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000005  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000
```

I see you have additional commits on [#7046 ](https://github.com/micropython/micropython/pull/7046) as I write this, so I'll pull and run against the latest updates and report back!

z7jzk, 2021-04-07:
When I implement `_IRQ_GET_SECRET` per the documentation at [Micropython uBluetooth Event-Handling](https://docs.micropython.org/en/latest/library/ubluetooth.html#event-handling), I get an error 

`NameError: local variable referenced before assignment`

```
elif event == _IRQ_GET_SECRET:
        # Return a stored secret.
        # If key is None, return the index'th value of this sec_type.
        # Otherwise return the corresponding value for this sec_type and key.
        sec_type, index, key = data
        return value         <~ Error is happening at this line
```

`value` is not defined per above, so I was wondering what exactly I should return?

z7jzk, 2021-04-13:
@dpgeorge thanks! I have been working through that example. I'm doing a little refactoring as the secrets are not being stored in the json file (cleaning this up). I'll have to follow up later tonight with where I'm at.

z7jzk, 2021-04-16:
@dpgeorge, maybe you can help me understand a few things as it relates to following the example at `examples/bluetooth/ble_bonding_peripheral.py`:

when this method is called on disconnect, no secrets are saved. When should I collect `sec_type`. `key`. and `value` to store the right values after bonding?
`def _save_secrets(self):
        try:
            with open("secrets.json", "w") as f:
                json_secrets = [
                    (sec_type, binascii.b2a_base64(key), binascii.b2a_base64(value))
                    for (sec_type, key), value in self._secrets.items()
                ]
                json.dump(json_secrets, f)
        except:
            print("failed to save secrets")`

It's not exactly clear to me what function call triggers this event type:
`elif event == _IRQ_SET_SECRET:
            print('set secret')
            sec_type, key, value = data
            key = sec_type, bytes(key)
            value = bytes(value) if value else None
            print("set secret:", key, value)
            if value is None:
                if key in self._secrets:
                    del self._secrets[key]
                    return True
                else:
                    return False
            else:
                self._secrets[key] = value
            return True`

I can't get the peripheral that I'm connecting with to read in the write's that I'm doing without first setting the peripheral into pairing mode each time.

z7jzk, 2021-04-24:
@dpgeorge, I'm sorry to be a continued pain, but I'm still struggling a bit to fully implement pairing on my device. I've referenced the example at `examples/bluetooth/ble_bonding_peripheral.py` and have a few questions yet. To add more context of the device I'm trying to setup, I'm planning to use my ESP32 as a Central device that will connect to a Peripheral device that requires a bonded connection.

1. When does the IRQ event `_IRQ_SET_SECRET` get called? Will `_IRQ_SET_SECRET` be called with the configuration I'm going after?
2. I've setup my code to use a hybrid from the `ble_bonding_peripheral.py` and `ble_simple_central.py` files; when `_save_secrets()` is called, the variables `sec_type`, `value`, and `key` are saved in the `secrets.json` file; however, I'm not sure when the appropriate time to set these variables are.
```
def _save_secrets(self):
        try:
            with open(self.secFile, 'w') as f:
                json_secrets = [
                    (sec_type, binascii.b2a_base64(key), binascii.b2a_base64(value))
                    for (sec_type, key), value in self._secrets.items()
                ]
                json.dump(json_secrets, f)
        except:
            print('failed to save secrets')
```

### Here is a full copy of the file I'm using for reference:
**_file I'm calling the class in bluetooth file from:_**
```
ble = bluetooth.BLE()
addrType=1
self.btc = btu.BLECentral(ble, self.bt_addr, self.bt_serv_uuid, self.bt_char_uuid)
self.btc.connect(addr_type=addrType, addr=self.bt_addr)
```

**_bluetooth file:_**
```
import bluetooth
import struct
import json
import binascii
from ble_advertising import advertising_payload
from ble_advertising import decode_services, decode_name
from micropython import const

_IRQ_CENTRAL_CONNECT = const(1)
_IRQ_CENTRAL_DISCONNECT = const(2)
_IRQ_GATTS_WRITE = const(3)
_IRQ_GATTS_READ_REQUEST = const(4)
_IRQ_SCAN_RESULT = const(5)
_IRQ_SCAN_DONE = const(6)
_IRQ_PERIPHERAL_CONNECT = const(7)
_IRQ_PERIPHERAL_DISCONNECT = const(8)
_IRQ_GATTC_SERVICE_RESULT = const(9)
_IRQ_GATTC_SERVICE_DONE = const(10)
_IRQ_GATTC_CHARACTERISTIC_RESULT = const(11)
_IRQ_GATTC_CHARACTERISTIC_DONE = const(12)
_IRQ_GATTC_DESCRIPTOR_RESULT = const(13)
_IRQ_GATTC_DESCRIPTOR_DONE = const(14)
_IRQ_GATTC_READ_RESULT = const(15)
_IRQ_GATTC_READ_DONE = const(16)
_IRQ_GATTC_WRITE_DONE = const(17)
_IRQ_GATTC_NOTIFY = const(18)
_IRQ_GATTC_INDICATE = const(19)
_IRQ_GATTS_INDICATE_DONE = const(20)
_IRQ_ENCRYPTION_UPDATE = const(28)
_IRQ_MTU_EXCHANGED = const(21)
_IRQ_GET_SECRET = const(29)
_IRQ_SET_SECRET = const(30)
_IRQ_PASSKEY_ACTION = const(31)

_ADV_IND = const(0x00)
_ADV_DIRECT_IND = const(0x01)
_ADV_SCAN_IND = const(0x02)
_ADV_NONCONN_IND = const(0x03)

_PASSKEY_ACTION_INPUT = const(2)
_PASSKEY_ACTION_DISP = const(3)
_PASSKEY_ACTION_NUMCMP = const(4)
[... 175 lines cut ...]
    def connect(self, addr_type=None, addr=None, callback=None):
        self._addr_type = addr_type or self._addr_type
        if self._addr_type == None:
            self._addr_type = 0
        addr = binascii.unhexlify(addr.strip().replace(':',''))
        self._addr = addr or self._addr
        if self._addr_type is None or self._addr is None:
            return False
        self._ble.gap_connect(self._addr_type, self._addr, 4000)
        return True

    # Disconnect from current device
    def disconnect(self):
        if not self._conn_handle:
            return
        self._ble.gap_disconnect(self._conn_handle)
        self._reset()

    # Send data over the UART
    def write(self, v, response=False):
        if not self.is_connected():
            print('not connected')
            return
        self._ble.gattc_write(self._conn_handle, self._rx_handle, v, 1 if response else 0)
```

chriscmorgan, 2021-09-27:
@z7jzk, was wondering if you where able to get this working or not?
I was looking to implement bonding for my esp32-c3 device but have been struggling so far.

z7jzk, 2021-11-07:
@chriscmorgan, sorry for the delay in my response. I ended up refactoring my entire project over to ESP-IDF directly. I struggled with older releases to get this to work, but was finally able to successfully pair and maintain the bond for repeated connections going forward starting with IDF version 4.3.1 about a month back. I suspect you may be able to get this to work with the latest firmware version on micropython.

dpgeorge, 2022-07-22:
Pairing and bonding is supported on esp32 since 5dbb822ca4a809ac5cb4513afb0411b4eb8dc3cf

closed 2022-07-22 by dpgeorge