micropython/micropython #7012: Micropython rp2 fails to build on windows systems.

WestfW, 2021-03-11:
When attempting to build micropython for rp2040 in a windows environment, the make/cmake environment generates a command line that exceeds the windows10/cmd.exe line length limit of 8191 characters.
This is because the RP2 SDK requires about 3k worth of C include paths, and the list of files compiled that is generated on a single command line is about 6k (depending on exact path names.)

Extensive discussion and whining here: https://www.raspberrypi.org/forums/viewtopic.php?f=144&t=306333

I'll add that compiling all the C files for micropython with a single command line seems ... dangerously likely not to scale well on other build platforms as well.

dpgeorge, 2021-03-11:
Do you have a suggestion how this could be fixed?

dhalbert, 2021-03-11:
We were seeing something similar in CircuitPython, and @jepler had this suggestion - I'm not sure it was tested yet, or if it's applicable here, but the basic idea is to use the `@filenames_file` feature of `gcc` to avoid putting all the flags and the filenames on the same line: https://gist.github.com/jepler/4d3687a746aa979a72984afc29852ee4.

From the gcc manpage:
```
 @file
           Read command-line options from file.  The options read are
           inserted in place of the original @file option.  If file does
           not exist, or cannot be read, then the option will be treated
           literally, and not removed.

           Options in file are separated by whitespace.  A whitespace
           character may be included in an option by surrounding the
           entire option in either single or double quotes.  Any
           character (including a backslash) may be included by
           prefixing the character to be included with a backslash.  The
           file may itself contain additional @file options; any such
           options will be processed recursively.
```

JohnieBraaf, 2021-08-19:
I'm running into the same issue while building stm32 ports.
@dhalbert did you find a workaround for building on Windows?

dhalbert, 2021-08-19:
Here is what we did in CircuitPython: https://github.com/adafruit/circuitpython/pull/4480.

Also we have a CI build that does a representative set of builds to make sure that the Windows build hasn't broken:https://github.com/adafruit/circuitpython/blob/main/.github/workflows/ports_windows.yml. But almost none of us build on Windows natively. Some users use WSL.

JohnieBraaf, 2021-08-19:
Thank you for the concrete pointers @dhalbert , I managed to resolve the issue with this method.

If this line:
https://github.com/micropython/micropython/blob/44818d1a35315c4dde34050eb0aac98866f9eb15/ports/stm32/Makefile#L619

Is replaced by this:
`$(Q)echo $(LDFLAGS) > $(BUILD)/ldflags.objs
$(Q)echo $(LDFLAGS_MOD) > $(BUILD)/ldflags_mod.objs
$(Q)echo $(LIBS) > $(BUILD)/libs.objs
$(Q)$(LD) @$(BUILD)/ldflags.objs -o $(1) $(2) @$(BUILD)/ldflags_mod.objs @$(BUILD)/libs.objs`

It works just fine and works around the windows max character limitation of console commands.

@dpgeorge , I can make a PR for this if that is ok for you. What are your thoughts?

jimmo, 2021-08-19:
Sounds like we might be seeing the same issue for makemoduledefs.py -- See #7687

JohnieBraaf, 2021-09-17:
I have found that the command line restriction can be worked around by using [MSYS2](https://www.msys2.org/)