micropython/micropython #7013: f.write() prints number bytes written in nightly

jdtsmith, 2021-03-11:
Today's nightly ESP 32 non-SPIRAM build:

```
>>> with open("file.txt","w") as f: 
...    for i in range(3):
...     f.write('Testing\n'*(i+1))
... 
8
16
24
```

dpgeorge, 2021-03-11:
That's expected behaviour.  At the REPL all results of executing a line a printed (but usually the result is `None` which is not printed).

dpgeorge, 2021-03-11:
Just to be clear, this is CPython behaviour:
```
$ python
Python 3.9.1 (default, Dec 13 2020, 11:55:53) 
[GCC 10.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> for i in range(4):
...     i
... 
0
1
2
3
>>> 
```

closed 2021-03-11 by dpgeorge

jdtsmith, 2021-03-13:
Interesting.  I'm suppose I'm used to ipython behavior, which only prints the _last_ item in a multi-line block by default. Thanks!