micropython/micropython #7026: rp2: KeyboardInterrupt thrown in 2nd thread instead of main

kevinkk525, 2021-03-13:
I remember having read about it somewhere but couldn't find it before opening this issue..

When using a thread on the pico, my KeyboardInterrupt is always thrown in that thread running on the 2nd core. This is problematic as it always stops my background library instead of whatever I just executed in the repl. The 2nd KeyboardInterrupt will get thrown in my repl on the 1st core but then I always have to reset the device or start the thread again.
In a running application this might not be a problem but for testing it's a nuisance.
Is there any easy way to change this behaviour? I'd say the thread on the 2nd core doesn't even need to catch a KeyboardInterrupt.

dpgeorge, 2021-03-13:
I'm not exactly certain, but I think CPython enforces that all signals are to be handled by the main thread, including KeyboardInterrupt.  That's obviously not happening in MicroPython so should be fixed.

kevinkk525, 2021-03-13:
https://docs.python.org/3.9/library/signal.html#signals-and-threads

> Python signal handlers are always executed in the main Python thread of the main interpreter, even if the signal was received in another thread.

You're right.

dpgeorge, 2021-05-08:
> Python signal handlers are always executed in the main Python thread of the main interpreter, even if the signal was received in another thread.

That's apparently only true if the `signal` module is available.  The caveats at the end of https://docs.python.org/3/library/_thread.html say this:

> Threads interact strangely with interrupts: the KeyboardInterrupt exception will be received by an arbitrary thread. (When the signal module is available, interrupts always go to the main thread.)

kevinkk525, 2021-05-10:
oh I didn't read that, should have thought about reading the low-level library. Afaik signal is not available on micropython and from the looks of it, it doesn't offer much for micropython either. 
So what's the conclusion? 
If it's easily achievable, processing the KeyboardInterrupt in the main thread would make development a lot easier, even if it's not completely CPython conform without the signal module.

dpgeorge, 2021-06-18:
Should be fixed by 259d9b69fe1f0e5c473bad85755d59233e0d182e

closed 2021-06-18 by dpgeorge