micropython/micropython #7031: rp2: uos.urandom() is missing.

peterhinch, 2021-03-14:
```python
MicroPython v1.14 on 2021-03-14; Raspberry Pi Pico with RP2040
Type "help()" for more information.
>>> import uos
>>> dir(uos)
['__class__', '__name__', 'remove', 'VfsFat', 'VfsLfs2', 'chdir', 'getcwd', 'ilistdir', 'listdir', 'mkdir', 'mount', 'rename', 'rmdir', 'stat', 'statvfs', 'umount', 'uname']
>>> 
```

teuler, 2021-03-14:
'random' seems to be implemented instead ...

dpgeorge, 2021-03-15:
The RP2040 doesn't have a good source of entropy (that I know of).  I tried using `rosc_hw->randombit` to generate the data for `urandom()` but it's not very random.

@lurch do you know if it's possible to generate true random numbers from the HW?  It might require feeding `rosc_hw->randombit` into a PRNG...

peterhinch, 2021-03-15:
Even a deterministic PRNG (with a warning in the docs) would be preferable to a missing method, which breaks code.

robert-hh, 2021-03-15:
An efficient method is feeding the rosc_hw->randombit output into a CRC. Still not very random, but well distributed. If I understand the literature right, you could also feed that bit into the yasmarang generator by XOR-ing the bit at this line instead of the 1. That has a way longer cycle than a CRC32.

`    yasmarang_dat ^= (char)yasmarang_pad ^ (yasmarang_d >> 8) ^ 1;`

lurch, 2021-03-15:
I asked some colleagues, and apparently this is "known behaviour": https://www.raspberrypi.org/forums/viewtopic.php?t=302960#p1814814

robert-hh, 2021-03-16:
Using PIO and DMA I routed the randombit output to a GPIO pin to take the spectrum. That is indeed not quite good. At room temp the prevailing frequencies are 2.5 and 4.5 MHz. Heating the device up, this pair of frequencies shifts to 6.7 and 8.5 MHz. At room temp, the lower frequency is about 10 dB below the higher. Cooling the device down results in the 2.4 MHz peak to drop further.
[image]
[image]
[image]

robert-hh, 2021-03-16:
To extend that further, here is the timing view. The frequency changes between the lower and higher domain regularly. 
[image]

So to get better random numbers, the sampling frequency must be substantially lower than the frequency of the ROSC. The samples shown in the link and also in the function rosc_random_u32() sample to fast. Some examples on the test below. I have taken 100k samples with different settings and plotted the histogram.
1. Sampling 8 bit in a byte as fast as possible:
[image]

2. Sampling 8 bit at a rate of 2 µs/bit
[image]

3. Sampling 32 bit at a rate of 2 µs/bit and folding them to 8 bit (all bytes XORed).
[image]

4. Sampling 32 bit at a rate of 2 µs/bit and shifting them through a 8 bit CRC.
[image]

It does not have to be 2µs/bit. That was just the shorted delay I could achieve in Python code with sleep_us. Tests with a sample rate of ~1 MHz, 32 samples/byte and the CRC folding also looked good. I did not do a standard RNG test yet. That requires MBytes of data. I have to set that up later.

peterhinch, 2021-03-17:
Have you tried a Von Neumann extractor, as per [the link](https://www.raspberrypi.org/forums/viewtopic.php?t=302960#p1814814) from @lurch?

robert-hh, 2021-03-17:
No, I did not. I looked at it and it did not seem better. It also requires in average 4 samples to retrieve one bit, and the algorithm is more complicated. The key to random quality is slow sampling. In the worst case, with very fast sampling the Von Neumann approach would catch every transition and produce a perfect 010101..... sequence. With the approach above, I ran a small data set of 10 MByte through the NIST 800-22 test suite, and it performed well. All tests passed. Over night I collected the full data set (135MByte) and as of now the sts test is running, up to this evening.
I have simplified my code a little bit to 8 samples/byte, by starting each byte with the previous value and thus mixing the history. The distribution looks fine, let's see what the statistical tests reveal. Should be slightly worse. The improvement is in speed getting ~120kByte/sec instead of ~30 kByte/sec.

robert-hh, 2021-03-17:
The 135MB sts test performed well. All test passed. The speed improved version with 10 MByte data also performed well with full test coverage, not worse as expected. And that code is definitely faster and simpler than the von Neumann approach. So I'll go with that one for a PR.

lurch, 2021-03-17:
> The key to random quality is slow sampling.

One of our hardware engineers says the other approach would be to increase the ROSC frequency significantly, but apparently that would increase the chip's power-draw.

robert-hh, 2021-03-17:
If you increase the frequency of the oscillator, the sampling frequency can be raised too. In any case the sampling frequency must be substantially lower than the ROSC frequency. And as long as no one complains about low speed of uos.urandom(), I see no need to change the ROSC setting. The low power consumption of the RP2040 is really a benefit.

dpgeorge, 2021-04-09:
Very nice investigation @robert-hh .

urandom is added in 6f06dcaee57cc6f36a4121b4797942200b2d7281

closed 2021-04-09 by dpgeorge