micropython/micropython #7032: rp2 SemVer x.y.patch zero

kevindawson, 2021-03-14:
rp2
after every install 
- rp2-pico-20210314-unstable-v1.14-**111**-gdcaf70257.uf2
```
>>> 
>>> import uos
>>> print(uos.uname().release)
1.14.0
>>> print(uos.uname().version)
v1.14 on 2021-03-14 (GNU 10.2.0 MinSizeRel)
>>> 
```
Over the last three weeks - uos.uname().release
MAJOR = 1
MINOR = 14
PATCH = **0**
I have been expecting the patch to match that of the uf2 file
Semantic Versioning Specification (SemVer)
MAJOR.MINOR.PATCH

dpgeorge, 2021-03-14:
MicroPython does not use strict semver.  If it did the version number corresponding to the above would be: `v1.14.0+111-gdcaf70257`

The `111` is the number of commits since `v1.14`.

kevindawson, 2021-03-15:
@dpgeorge 

Thanks for that nugget of info, in that case:
- Then number of commits is approximately equal to semver patch

if you made it so
- patch would have meaning
- you could request ```uos.uname().release``` for new issues

I am thinking rp2 **array** here, missing in 1.14.0 - fixed in 1.14.0
I think you should treat commits and patch as the same

dpgeorge, 2021-03-15:
> I think you should treat commits and patch as the same

There are cases where there's a release that does include a patch, say, 1.14.1.  Then there would be versions like `v1.14.1-123-gabcdef` (ie 123 commits since release 1.14.1).

kevindawson, 2021-03-16:
@dpgeorge 
silly question, sorry to bug you about this, how do I ask micropython what *.uf2 it was extracted from Please.

dpgeorge, 2021-03-16:
> how do I ask micropython what *.uf2 it was extracted from

It does not know what uf2 it was packaged in to.  The best it knows is `os.uname().version`.  What additional information do you need?

kevindawson, 2021-03-16:
@dpgeorge 
Thank you for the prompt reply
Sorry to be so woolly headed - but I have an issue with some of my code that likes different `.uf2` and I am trying to track my issue before raising an issue - as the fault is probably me ;)
I will just have to go back to old school of a list on paper of `.uf2` <-> date
my bad ;)

dpgeorge, 2021-03-16:
The result of `os.uname().version` should be enough to pin-point a particular commit to know where the problem was introduced.  That is more robust than the uf2 date.

It might be easier if you state the bug/issue that you found?

kevindawson, 2021-03-17:
at the current time I believe the issue is all my own doing, just trying to eliminate micropython, nothing more

I am happy for this to close

So long and Thanks for all the Fish!

closed 2021-03-17 by kevindawson