micropython/micropython #7034: Porting MicroPython to Renesas RA4M1 and RA6M2

S-Nakamura-R, 2021-03-15:
Hi, 
I am Saori Nakamura from Renesas Electronics.

We ported MicroPython to our Cortex-M4 based Microcontroller, RA4M1 and RA6M2.
Now we are planning to submit pull request.
If there is something we should consider, would you please let us know?

We confirmed Contributor Guideline and #5449.
In our case, the answers for the items listed in #5449 are as followings.

1. are the chips widely available to the public and will they be for the foreseeable future?
=> Yes. 
Access the pages and click “Buy / Quote” of “Product Options”, you can find many distributors.
   RA4M1:
https://www.renesas.com/us/en/products/microcontrollers-microprocessors/ra-cortex-m-mcus/ra4m1-32-bit-microcontrollers-48mhz-arm-cortex-m4-and-lcd-controller-and-cap-touch-hmi
RA6M2: 
https://www.renesas.com/us/en/products/microcontrollers-microprocessors/ra-cortex-m-mcus/ra6m2-32-bit-microcontrollers-120mhz-medium-size-memory-integration-and-ethernet
And, “RA” series is part of the Renesas Product Longevity Program (PLP).
     https://www.renesas.com/us/en/support/product-longevity-program-plp

2. are development boards easily available?
=> Yes. RA is already widely available and will be even more in the future.
 RA4M1 Clicker board:
https://www.mikroe.com/ra4m1-clicker
Also, the boards are easily available from Digi-key, Mouser, and etc.
 EK-RA6M2:
https://www.renesas.com/us/en/buy-sample/check-inventory/result?show_price=1&partno=RTK7EKA6M2S00001BU&exact=1
The boards are easily available from many distributors.

3. is the toolchain open source (or at least free to obtain)?
=> Yes. GCC is used and it is available .
https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads

4. are there plans to support multiple MCUs/SoCs?
=> Yes, it is our intention to extend MiroPython support to multiple groups within the RA MCU lineup, and also potentially to other Renesas MCU families.

5. will there be someone around for the long term to maintain this port?
=> Yes, it is our intention to offer long term maintenance of this port, in line with Renesas’ usual policy of customer care.

6. the license (and copyright) must be appropriate.
=> Yes, the license follows https://docs.micropython.org/en/latest/license.html

7. support for a new device must be sustainable in the long term (that's a general remark).
=> Yes, we fully agree, please refer to 5.

8. a new port should ideally not touch any files in any directory except the new port directory (and any new submodules), so fixes/updates to the core (py/, extmod/, etc) should be in separate PR's
=> Yes.

9. the new port name should be generic enough in scope to encompass new MCUs from the same family in the future
=> Yes, we fully agree and this is our intention.

The ported codes follow the Code Conventions and we applied the test suite included this package and confirmed the results.

stinos, 2021-03-15:
Answering 'yes' on all points is a good start for sure :)

@dpgeorge perhaps just add https://github.com/micropython/micropython/pull/5449#issuecomment-714454904 to Contributor Guidelines?

S-Nakamura-R, 2021-03-23:
@stinos 
Thanks!  I created PR.

chrismas9, 2021-03-23:
Great to see this port. The RA4M1 is the first 5V MCU running MicroPython. 5V is still used in noisy automotive and industrial applications and can drive power MOSFETs directly.

Meekdai, 2022-04-15:
Great, I hope to have the opportunity to use Renesas MCU running micropython in the future.

dpgeorge, 2022-04-29:
The renesas-ra port was added in 0f57ccf7f2d396317b41cd90e0599fc7d4d182d8

closed 2022-04-29 by dpgeorge