micropython/micropython #7038: BLE and HTTPS request issue (urequests and ubluetooth library)

EstebanHardware, 2021-03-15:
Hi community!

As a first time person, I am new here and only a couple of months into working with ESP32. Therefore, any help from you will be appreciated.

I am working on a kind of BLE / WiFi gateway to detect some beacons and send them over the internet.

I only have a problem when I activate the BLE and try to send a request using the https protocol. (The problem does not occur with the http protocol)

Some clues:
-If I make an http request I can have the BLE active without any problem
-If I make an https request without active BLE, the requests work fine
-If I make an https request with BLE active, an OSError: [Errno 12] ENOMEM occurs. The problem appears when the urequests library invokes the ussl.wrapsocked function on line 60
-The micropython.mem_info(1) shows enough free space RAM memory

The development board is a generic 4MB ESP-WROOM-32.
Currently running firmware esp32-idf3-20200902-v1.13.bin, but I tried with:
-esp32-20210310-unstable-v1.14-84-g59a129f22.bin
-esp32-idf3-20180511-v1.9.4.bin
-esp32-idf3-20190529-v1.11.bin
-esp32-idf3-20191220-v1.12.bin
-esp32-idf3-20210202-v1.14.bin
-esp32-idf4-20210202-v1.14.bin
And also with my own build, but the same problem occurs.

Here the example code:
`
import network
import time
import urequests
import micropython
import webrepl
import gc
import ubluetooth

#GARBAGE COLLECTION
gc.collect()
gc.threshold(gc.mem_free() // 4 + gc.mem_alloc())
gc.enable()

#CONNECT TO WIFI
wlan = network.WLAN(network.STA_IF)
wlan.active(True)
print('--WLAN INIT--')
wlan.connect("Psarocolius" , "Oropendola2021")

#WAIT WHILE CONECTION IS DONE
var = 0
while (not wlan.isconnected() and var<10):
var += 1
print("--WLAN CONNETION TRY ",var)
time.sleep(1)
if wlan.isconnected():
print('--WLAN SERVICE ON-- ', wlan.ifconfig())

#HTTPS URL TO POST REQUEST
url = 'https://www.google.com/'

#ACTIVE BLUETOOTH
ble = ubluetooth.BLE()
ble.active(True)

#DESACTIVATE BLUETOOTH
#time.sleep(1)
#ble.active(False)

#MEM INFO
gc.mem_free()
#micropython.mem_info(1)

#POST REQUEST
respon = urequests.post(url)
print('Status Code: ',respon.status_code)

#MEM INFO
#micropython.mem_info(1)
gc.mem_free()
`

cookejames, 2021-04-08:
I'm having similar issues though it is with an MQTT over SSL and bluetooth resulting in ENOMEM even when gc.mem_free() reports 60KB memory free.

dpgeorge, 2021-04-13:
This is most likely due to the FreeRTOS heap running out of memory, not the MicroPython heap (ENOMEM means an OS memory error, a Python memory error is indicated by `MemoryError` exception).

The solution would be to use less memory for the MicroPython heap.  A way to do this is provided in #6785, but that is not yet merged to master.

cookejames, 2021-04-13:
> This is most likely due to the FreeRTOS heap running out of memory, not the MicroPython heap (ENOMEM means an OS memory error, a Python memory error is indicated by `MemoryError` exception).
>
> The [...]

Thanks, it seems plausible as I've rewritten my code in C++ and not had any problems. Is it possible to see the amount free on the FreeRTOS heap?

cookejames, 2021-04-13:
> This is most likely due to the FreeRTOS heap running out of memory, not the MicroPython heap (ENOMEM means an OS memory error, a Python memory error is indicated by `MemoryError` exception).
>
> The [...]

If I get time in the next couple of days I'll build the PR you mentioned and see if that helps.

dpgeorge, 2021-04-13:
> Is it possible to see the amount free on the FreeRTOS heap?

See the `esp32.idf_heap_info(...)` function.

enesbcs, 2021-05-09:
MicroPython v1.15-64-g1e2f0d280-dirty

SSL connection response:
connection failed [Errno 12] ENOMEM

import esp32
esp32.idf_heap_info(esp32.HEAP_DATA)
[(240, 0, 0, 0)]

Its not very informative.

enesbcs, 2021-05-14:
Tried the (micropython) heap "minus memory trick" in micropython main.c but still got ENOMEM (espidf) at SSL connection.

noodlemind, 2021-09-28:
The issue doesn't seem to be related to the memory. Validated the available memory space before and after the each step of the program execution.