micropython/micropython #7039: W5xx (wiznet5k) latency in STM32 port

urielka, 2021-03-15:
I opened a forum question with more details:
https://forum.micropython.org/viewtopic.php?f=12&t=10053&p=56073#p56073

What I found there is that the wiznet5k driver and integration in the STM32 port works with a polling mechanism for LWiP that polls the wiznet chip every 128ms (per the comment in https://github.com/micropython/micropython/blob/master/ports/stm32/modnetwork.c#L52):
`#define LWIP_TICK(tick) (((tick) & ~(SYSTICK_DISPATCH_NUM_SLOTS - 1) & 0x7f) == 0)` 

This causes reading from socket or answering ping requests to have a latency from 0-128ms that varies according to when you issue the recv from the socket compared to the poll time.

I believe that this behavior is not expected and not documented anywhere (took me a day to figure out).

I suggest the following:
1. To document this in the network/wiznet module for future users
2. Add a configuration option to allow setting different polling interval, in my use case this latency is not suitable for me
3. (Harder) add support for wiznet5k interrupt pin - so instead of polling the same function will be scheduled via an interrupt handler on that pin

jonnor, 2024-09-13:
This polling still seems to be present. Code currently in https://github.com/micropython/micropython/blob/f1bdac375240941edfc0aa04b646dc3c53d6b371/ports/stm32/mpnetworkport.c#L52

andrewleech, 2024-09-14:
When the driver was migrated to the extmod version support for the interrupt pin was added, which I believe should work for stm32 as well. 
However it's optional / only used if the pin is configured, otherwise it fails back to the polling functionality.
It was recently noticed that the documentation for this never got updated to cover the interrupt pin however... that probably should be raised as its own issue so it's not completely forgotten.