micropython/micropython #7043: help with low free mem after recompile.

kindmartin, 2021-03-16:
Hi need a probably silly help as I lost the ubuntu vm that had the esp32 toolchain so create a new one. 
With all compiling options I had, I used to have >100kB of micropython.mem_info() after boot. now no more than 60kB. 

before (old compilation ) and after (today)

what could be the config change to produce max ram for the same ESP32 S module?

I think the included modules at port/esp32/modules were the same.

what is the normal free mem available for the module?

thanks in advance.

M

before:  ----->>>>

MicroPython v1.13-186-g5a7027915-dirty on 2020-12-10; ESP32 module with ESP32
Type "help()" for more information.
>>> import gc
>>> gc.mem_free()
91088
>>> micropython.mem_info(1)
stack: 720 out of 15360
GC: total: 111168, used: 20304, free: 90864
 No. of 1-blocks: 299, 2-blocks: 54, max blk sz: 43, max free sz: 3400
GC memory layout; from 3ffe4db0:
00000: MDShhBMhSDhhhh=Bhh=================hh=======h=======h=hh========
00400: =======ShSShTMhBDSh==Sh=========h===============hSSShhhShBMDhShS
00800: h=h=======h=====BTT=BTh=hBhhhBDh===============STLhBh==T=Dh=BDBD
00c00: Sh==hTh===BDhTBBSh=======h=h=======h===hhBhSSh=h=======SSSh=h==T
01000: B=BBDh==h===h==h==========h==h==TB=BhBDBBBBBBBSh==h=======SShBBh
01400: ===BBBBBh=h==hSSh====================h====Bhhh=hDDBh=BBBBBh===TB
01800: =Bh=====h==============h==============h====h==BT=B=BTB=BTB=BBh==
01c00: BBBBTB=BBBBBBBBhhhhBBhh=hBh=h=h=B..h....h=h=........Sh==========
02000: ===================..............................h==============
02400: ======================...h=======...............................
02800: ........h.....hh...h.....hSh...h..hh..h=........................
       (2 lines all free)
03400: .........h==................................h=======.........h..
03800: ............Sh====================.........h==..................
03c00: ........Sh.........................h==..........................
04000: ...Sh==....................Sh...................................
04400: ................Sh=........SSh.........h........................
04800: ......Sh.........h=======.........Sh=.........Sh=...............
04c00: Sh=.........Sh==Sh=..............................Sh=............
05000: ................S......hhh...hh=...hh.....................h.....
05400: ....Sh==........SSh=..................h=........................
05800: .........F.........FF.........F................Sh=..............
05c00: ............Sh.........h=======.................................
06000: ..........................Sh................................Sh..
06400: ............................Sh==.......................Sh=......
06800: ......................................................h=======..
06c00: .......................Sh.............Sh=............Sh.........
07000: .......................................................Sh=......
07400: ...............Sh.......................Sh=............Sh.......
07800: ........................................Sh=............Sh.......
07c00: ..........................................................Sh....
08000: .....Sh==.......................Sh.................Sh===........
08400: .............................Sh==......................Sh==.....
08800: ................................................................
08c00: ...................................................Sh.........h=
09000: ======.............................Sh==.........................
09400: .........hhh====............h=======.................Sh===......
09800: ........................................Sh==....................
09c00: ...Sh==...............Sh=...........Sh=.........................
0a000: .Sh==....................Sh............Sh===....................
0a400: .Sh.........Sh=.........................Sh===...................
0a800: .....................Sh=.............Sh=...............h=======.
0ac00: .......SSh=.........h=.............Sh...........................
0b000: ...........Sh=....................................hh=...hhh...hh
0b400: ...hhh...hhh...hh=...hhh...hhhhhhh...h=...hhh...hhh...hhh...h...
0b800: ..h.....hh..................h...................................
       (2 lines all free)
0c400: .....................................h==========================
0c800: ================h=======h===h==h==========h============h========
0cc00: ====h===========h===h==========h==========h=====================
0d000: =================h===h=====h============h======================h
0d400: ================h===========================h========h==========
0d800: =================h=====h==h=======h======h===================h==
0dc00: =========h====h====h==h=====....................................
       (52 lines all free)
1b000: ....................................

AFTER RE build (NOW, today)

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:5356
ho 0 tail 12 room 4
load:0x40078000,len:13716
load:0x40080400,len:3388
entry 0x40080634
Performing initial setup
MicroPython v1.14-118-gcb68a5741-dirty on 2021-03-16; BikeWatch module with ESP32
Type "help()" for more information.
>>> import gc
>>> gc.mem_free()
62240
>>> micropython.mem_info()
stack: 704 out of 15360
GC: total: 64000, used: 4736, free: 59264
 No. of 1-blocks: 19, 2-blocks: 11, max blk sz: 41, max free sz: 523
>>> micropython.mem_info(1)
stack: 720 out of 15360
GC: total: 64000, used: 4832, free: 59168
 No. of 1-blocks: 21, 2-blocks: 13, max blk sz: 41, max free sz: 523
GC memory layout; from 3ffcf170:
00000: h=hhhBMh=DBh=h=hhh=================hh=======h=======h=hBh=hB..h=
00400: h.h=.....h=.TMh=D.......................h====.h=.h.......h=h...S
00800: h=h=======h=====......h========.................................
00c00: .................h=======...........................h==========.
01000: .....h========h===============....................h=======......
       (7 lines all free)
03000: ............................................h=======............
03400: ....................h====================.......................
03800: ................................................................
03c00: ..........................................h=======..............
       (3 lines all free)
04c00: ..................................................h=======......
       (6 lines all free)
06800: .............h=======...........................................
       (7 lines all free)
08800: ................................h=======........................
       (2 lines all free)
09400: ..................................h=======......................
       (7 lines all free)
0b400: .......h=======.................................................
       (2 lines all free)
0c000: .h=======h========================================..............
0c400: ................................................................
0c800: ....h======.....................................................
       (4 lines all free)
0dc00: ...............................................................h
0e000: =======.........................................................
       (5 lines all free)
0f800: ................................
h=hhhBMh=DBh=h=hhh=================hh=======h=======h=hh========
00400: =======hBhhhBDhBhBhh=h=BhBhBhBh=h=hBhBhh=h=hhh=BhBhhh=h=h=hBShTh
00800: =h=h==hh=BhBh=hh==hBh=hhh=Bh=Bh=hh=h=h=Bh=hBShhTh=h===hh=h=B..h=
00c00: h.........h=....................................h=======h=====..
       (58 lines all free)
0f800: ................................

dpgeorge, 2021-03-16:
What IDF version did you compile with?  Try IDF v4.2.

kindmartin, 2021-03-17:
hey Damien, thanks your workload is incredible, I would like to achieve 1% of dealing with leading edge particle physics and lead a great project like this one, how do you have time to respond my probably silly question? awesome!

I followed each step mentioned at https://github.com/micropython/micropython/tree/master/ports/esp32 and also tried https://lemariva.com/blog/2020/03/tutorial-getting-started-micropython-v20

today Ive tried with IDF 4.2 and 4.1.1 and the mentioned v4.0.2
git clone -b v4.1.1 --recursive https://github.com/espressif/esp-idf.git

running on ubuntu 19.10.

any step by step tested guide you can point to follow?

thanks all.

dpgeorge, 2021-03-17:
> today Ive tried with IDF 4.2 and 4.1.1 and the mentioned v4.0.2

So with these versions do you still get only 64000 heap?  Did you try running official binaries from https://micropython.org/download?

kindmartin, 2021-03-17:
not really interested  in binaries as I need to include some modules, not big ones, like a agps lib and wifi json confif file, all less than 10k at .py by size, so mpy would be less.
but I will try tonight those binaries.

thanks,

M

kindmartin, 2021-03-17:
Just tried esp32-20210317-unstable-v1.14-119-g6e5aea08a.bin and esp32-idf4-20210202-v1.14.bin. both are ok, so there's something configured in my setup that makes this low mem/heap.
---->>>

MPY: soft reboot
MicroPython v1.14 on 2021-02-02; ESP32 module with ESP32
Type "help()" for more information.
>>> import gc
>>> gc.mem_free()
109872
>>> 

so physical module its ok, it must be something in my make config.

dpgeorge, 2021-03-17:
Try building it locally but with all your mods disabled, ie just build standard MicroPython esp32 using your Ubuntu set up.

kindmartin, 2021-03-17:
tried also that, no including any additional module. but I ve got  again just 60k free heap.

coping those modules to esp32 root running last stable 1.14 works, so for the moment, so I can go on. it would be genial if someone have a working VM with a working toolchain to ftp me, just not to deal fixing my current toolchain setup.

thanks for all, support included.

}M

>>> import reset
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5148
load:0x40078000,len:12880
load:0x40080400,len:3484
entry 0x40080630
bootStatus:   / bootCondition 
booting main code
machine.wake_reason() :: 0
machine.wake_reason() == 0 --> PowerOn()

--->> press button p34 to set next boot to Armed from PowerOn

MicroPython v1.14 on 2021-02-02; ESP32 module with ESP32
Type "help()" for more information.
>>> gc.
__class__       __name__        collect         disable
enable          isenabled       mem_alloc       mem_free
threshold
>>> gc.mem_free()
83584
>>> import os
>>> os.listdir()
['bikewatch.py', 'boot.py', 'ftp_thread.py', 'main.py', 'net0.py', 'reset.py', 'uping.py', 'urequests.py', 'wifis.py']
>>>

kindmartin, 2021-03-17:
I tried all once again, and solved. sorry if I took time.
if helps someone I deleted all, reboot and from home did:

mkdir esp
cd esp/
git clone -b v4.2 --recursive https://github.com/espressif/esp-idf.git
cd esp-idf/
./install.sh
. ./export.sh 
cd ..
git clone --recursive https://github.com/micropython/micropython.git
cd micropython
make -C mpy-cross
cd ports/esp32
make -j8 submodules
make -j8
#connect the usb serial to that A.I. ESP32S soc, then boot on bootloader
make erase
make deploy
 
#--->and voila...

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:5608
load:0x40078000,len:12696
load:0x40080400,len:4292
entry 0x400806b0
Performing initial setup
MicroPython v1.14-119-g6e5aea08a on 2021-03-16; ESP32 module with ESP32
Type "help()" for more information.
>>> import gc
>>> gc.mem_free()
109408

closed 2021-03-17 by kindmartin