micropython/micropython #7051: sdcard.py - OSError: timeout waiting for response

JurijOrlow, 2021-03-18:
I am using 8 GB microSD card, format FAT32. During mount code detects it as v2 card, but gives "OSError: timeout waiting for response". Tried to format as 4 GB FAT, did not work.

sgbaird, 2022-11-03:
Any status update on this one? I'm also running into this issue on the Pico W using https://github.com/micropython/micropython/blob/v1.19.1/drivers/sdcard/sdcard.py (https://github.com/micropython/micropython/issues/8942). Mine is also formatted as FAT32 (Samsung 32 GB Evo Plus).

```python
Traceback (most recent call last):
  File "<stdin>", line 22, in <module>
  File "/lib/sdcard/sdcard.py", line 59, in __init__
  File "/lib/sdcard/sdcard.py", line 93, in init_card
  File "/lib/sdcard/sdcard.py", line 152, in init_card_v2
OSError: timeout waiting for v2 card
```

jonnor, 2024-09-08:
The linked thread seems to have a bunch of people who got their cards working. So check that for tips&tricks. And a few that did not =/

VinceParsons, 2026-10-17:
If anyone here is still seeing `OSError: timeout waiting for response` on reads, there's a possible fix in micropython/micropython-lib#1166.

The current `sdcard.py` releases chip select between the blocks of a multi-block read/write. Some cards lose bit alignment across that release, and the data token then arrives shifted (`0xe0` instead of `0xfe`), so the driver times out. Adafruit made the same fix in their copy of this driver because Samsung EVO cards failed without it, which matches the EVO Plus report above.

To try it, replace your `sdcard.py` with this one (drop-in, same API):
https://raw.githubusercontent.com/VinceParsons/micropython-lib/sdcard-multiblock-cs/micropython/drivers/storage/sdcard/sdcard.py

If it fixes (or doesn't fix) your card, a comment on micropython/micropython-lib#1166 with the card make/model and board would help get it reviewed.