micropython/micropython #7053: b1d08726ee,  v1.10 \py\objtype.c  'MP_QSTR___int__': undeclared identifier

Zitt, 2021-03-19:
Unable to move to later builds of micropython in the UEFI implementation due to code commited #b1d0872 . V1.9.4 was used in UEFI port at:
https://github.com/tianocore/edk2-staging/tree/MicroPythonTestFramework/MicroPythonPkg

Compiles fail:
MicroPythonPkg\MicroPython\py\objtype.c(379): error C2065: 'MP_QSTR___int__': undeclared identifier

https://github.com/micropython/micropython/commit/b1d08726ee#r48434311

dpgeorge, 2021-04-13:
In the build, how are the qstr definitions generated?  Are they generated automatically or manually/by hand?  You may need to add `MP_QSTR___int__` to qstrdefsport.h.

jonnor, 2024-09-13:
Unfortunately, this issue does not have enough information to debug/reproduce. If the problem is still relevant on a recent MicroPython version, please provide the requested details and reopen the issue.

closed 2024-09-13 by jonnor

Zitt, 2024-10-18:
I'm getting the filling this was dismissed out of spite without actually looking at it.

Did you download and setup the UEFI build environment?
Did you attempt to move past v1.9.4 of the micropython source?
Were you able to compile without errors?

dpgeorge, 2024-10-18:
This was not closed out of spite, but instead with the aim to try and clear the backlog.

A request for information (see https://github.com/micropython/micropython/issues/7053#issuecomment-818764009) was not provided, so it was fair to assume that this could be closed.

Looking at the UEFI port, indeed the qstrs are hard coded in this file: https://github.com/tianocore/edk2-staging/blob/MicroPythonTestFramework/MicroPythonPkg/MicroPythonDxe/Uefi/genhdr/qstrdefs.generated.h .  That file needs to be regenerated when you update the MicroPython version.  As @stinos pointed out in a thread linked above, there's a tool in that UEFI port to do this.  You need to follow the README for that tool.

Also, if you update to a very recent MicroPython version, that tool needs updating itself, because qstr generation has changed recently.