micropython/micropython #7060: BLE ESP32, functionnal, doc and examples issues

EddieParis, 2021-03-21:
BLE is not very easy, and on ESP32 it is even worse.

In the documentation for bluetooth module, for config function, it is not specified that some parameters are not available for ESP 32 (bonding related thing). So is it in the example directory, the bonding one cannot work on ESP32 since it is using some of the not supported parameters.

Also, I went crazy for hours before realizing what was wrong around my newly implemented battery service. The UUIDs are either a string (long UUID) or an integer (short UUID).
My problem was that I used a string for short UUID (e.g. '180F' for battery service instead of 0x180F). The point is that it is accepted as an UUID, but then it was making the peer app crashing !
Why are the 4 bytes long UUID as string accepted ? I saw in the c code that it is accepted only in a kind of copy constructor, but for me these 4 bytes UUID can't exist and then they should not be accepted.

Also, somewhere between v1.12 and 1.13 the IRQ event codes have changed. This is a major break of API. I know it is still under development API, but a small note would have been appreciated. My old APP was not working anymore without any clue.

After agreement, I can do the changes and submit a pull-request.

Manudulux, 2021-03-22:
I would so much like to have an example working on ESP32 in  a central role reading data from a peripheral reading data from a bluetooth UID from a phone (like Komoot see :  https://www.komoot.com/b2b/connect#bleconnect ). I manage to scan, find the service, find the characteristics but I cannot read the service data. Either I am missing something or I am facing similar issues as Eddieparis. 
I understand this is still development stage but documentation specific to ESP32 would be a great bonus.

dpgeorge, 2021-04-13:
> Also, somewhere between v1.12 and 1.13 the IRQ event codes have changed. This is a major break of API. I know it is still under development API, but a small note would have been appreciated.

This change was mentioned in the change-log and release notes.  See eg https://micropython.org/resources/micropython-ChangeLog.txt, under release v1.13.

> Why are the 4 bytes long UUID as string accepted ?

It's accepted in case you have a binary representation of the UUID.  So the following are equivalent:
- `UUID(0x180f)`
- `UUID(b'\x0f\x18')`

> I would so much like to have an example working on ESP32

There are examples in the `examples/bluetooth/` directory.  They are intended to work on all ports that support BLE (ie they are not specific to esp32 but should work on esp32).

jonnor, 2024-09-13:
Most of the questions here seem to have been addressed. Can this issue be closed?

closed 2025-02-04 by projectgus