micropython/micropython #7065: Trivial issue:  1ej returns 1j (expected a Syntax Error or similar...)

ddiminnie, 2021-03-23:
As the title suggests, this really is a trivial item.  (We happened to notice it in our MicroPython fork, and thought we'd pass it on.)
(In case it matters, I'm running a 64-bit release build of MicroPython 1.14, built from code pulled from GitHub at 14:42 CDT on 23 March 2021.)

Within the REPL, entering 1ej returns 1j, rather than throwing a SyntaxError (or similar):
```
>>> 1ej
1j
```

For reference, CPython 3.7.5 throws a SyntaxError here:
```
>>> 1ej
  File "<stdin>", line 1
    1ej
      ^
SyntaxError: invalid syntax
```

ddiminnie, 2021-03-24:
Naturally, this behavior is not limited to the example above.  Rather, any time a 'partial "calculator scientific notation" expression' is entered before the 'j' symbol we see similar outcomes:
```
>>> 1.ej
1j
>>> 1.23456ej
1.23456j
>>> 1.23456e-j
1.23456j
>>> 1.23456e+j
1.23456j
```
(etc.)
We fixed this behavior in our own fork by introducing a new constant, 'PARSE_DEC_IN_EXP_DIG' to the enum 'parse_dec_in_t' (parsenum.c) to denote whether a digit was seen after the 'e' symbol, and setting/tracking the value at appropriate points in the state machine used by mp_parse_num_decimal().  However, I'm not convinced this solution is the most elegant in the world...

dpgeorge, 2021-04-14:
Thanks for the report.  Indeed this is not a particularly important thing to fix, I don't think there are any cases that one can enter a value which would be converted to something unexpected (in all the cases above there is just no exponent, as expected).

If it can be fixed with very minimal code-size change (say 8 bytes on ARM Thumb) then would be worth doing.

ddiminnie, 2021-04-14:
As I mentioned, we fixed it in our own port... but that fix is a bit clunky, so I wouldn't recommend it for a wider release (unless you have the code space to spare)... and it's a rather trivial issue.