micropython/micropython #7066: Slight inaccuracy in decimal formatting of floating-point values

ddiminnie, 2021-03-23:
This is essentially the inverse of the problem addressed in #6024 (which deals with parsing a stream of decimal values into floating-point values), and the behavior cited below probably does not matter for the vast majority of MicroPython's main applications, but we'll mention it in passing on the off chance that someone has a reasonably clean, tight fix for the issue:

When formatting floating-point numbers for display, the displayed value is not always the closest decimal value to the underlying floating-point value.

This behavior is most noticeable when working with floating-point integers (which can be stored exactly in IEEE-754 binary64 values).  The exact values that exhibit this behavior probably varies by port.  On a Windows 64-bit build of MicroPython 1.14 there are 3325 integers in the range 0., 9999. that are displayed "incorrectly":
```
MicroPython v1.14 on 2021-03-23; win32 version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> 41.
41.00000000000001
>>> 46.
46.00000000000001
>>> 9996.
9996.000000000002
>>> 9998.
9998.000000000001
>>> 9999.
9999.000000000002
```
A quick peek at the internal values for these floats shows that they have been stored accurately, e.g.
41. is stored as 0x1.4800000000000p+5, as expected, so the issue that #6024 is attempting to address is not at play here.  Rather, it's the conversion from (binary64) floating-point values to decimal equivalents for display that appears to be at fault.

dpgeorge, 2021-04-14:
Thanks for the report.  Yes this is indeed an issue, which is due to the complexity of formatting floats values for printing, and MicroPython tries to do a best-effort job in minimal code.

Note that most float tests work around this issue by printing only a limited number of decimal digits.

Would be good to fix this but I think it's a lot of work.

ddiminnie, 2021-04-14:
Understood.  I don't think this is a tremendous priority, but I thought you'd at least want to know.  😄

ddiminnie, 2022-05-23:
Incidentally, should a tight, accurate algorithm for producing correctly-rounded decimal floats from binary64 (and binary32) values come around, we probably ought to bump the precision in float_print (objfloat.c) to '17' for binary64 and '9' for binary32 values.  (These are the minimum number of decimal digits required to uniquely represent each binary value (whether binary64 or binary32) in decimal form.  (Proof left as an exercise for the reader.))

In that case the internal float equivalent to (e.g.) 0x1.0000000000001p+0 would be represented as 1.0000000000000002 and 0x1.0000000000002p+0 would be represented as 1.0000000000000004.  In the current MicroPython implementation both are represented as 1.0.

Correct float formatting is not a big deal for robotics or wearable electronics (which rely on the internal float values), but as python (and MicroPython in particular) is emerging in the math classroom this may become a bigger issue...

... but not currently enough to drop everything and pursue it as a priority.

dpwe, 2022-07-28:
Fixed in #8905

dpgeorge, 2022-07-28:
Yes this is indeed fixed, at least for integer values (represented as floats).

> Incidentally, should a tight, accurate algorithm for producing correctly-rounded decimal floats from binary64 (and binary32) values come around, we probably ought to bump the precision in float_print (objfloat.c) to '17' for binary64 and '9' for binary32 values

Yes I agree.  But we still don't have such an algorithm that works correctly for all floats, so printing 17 (or 9) decimals still gives a false sense of accuracy (and the decimal value printed does not exactly represent the internal stored value).

closed 2022-07-28 by dpgeorge

ddiminnie, 2022-07-28:
The use of 17 digits for formatting floats is not for accuracy (indeed, significantly more digits are required to represent the exact internal value); rather, it is for uniqueness - each internal float can be uniquely represented by a 17-digit decimal number.

The scenario where this becomes important is in classroom use of the REPL, where students/teachers paste previously-displayed results into later calculations (the same issue is present on modern graphing and scientific calculators).  By using only 16 digits, a value different from the previous result may potentially be used in future calculations (because 16 digit decimal representations cannot uniquely represent all the internal floating-point values).

It's a subtle point, and should anyone wish to pursue this it should be handled in a separate issue.  I just wanted to plant the seed... :-)