micropython/micropython #7068: ESP32 Quickref pin guidance should be improved

mattytrentini, 2021-03-25:
As discussed in the forum post [ESP-32 WROOM discrepancy in quickref page](https://forum.micropython.org/viewtopic.php?f=18&t=10076), the [ESP32 quickref documentation](https://docs.micropython.org/en/latest/esp32/quickref.html#pins-and-gpio) should be improved in the guidance given for pin usage. 

In particular, some of the different models have different pin restrictions. Need to at least consider the WROOM, WROVER and PICO D4.

(This can be assigned to me)

UnexpectedMaker, 2021-04-26:
Should also add an S2 pin guide while this is happening as the pins are quite different.

jonnor, 2024-09-13:
Is this still relevant @mattytrentini  ? Is there perhaps some official EspressIf doc we can refer to? Maintaining our own hardware-specific tends to become outdated real fast.