micropython/micropython #7073: rp2: make fail CMake Error: The source directory "..." does not exist.

maxpromer, 2021-03-26:
Hello !, I clone micropython and install follow step at [1.3. Building MicroPython From Source](https://datasheets.raspberrypi.org/pico/raspberry-pi-pico-python-sdk.pdf) but after i run `make` i got this

```
max@MAXNB:~/esp/micropython/ports/rp2$ ls
CMakeLists.txt  machine_i2c.c    machine_uart.c  modmachine.c  moduos.c        mpthreadport.c  tusb_config.h
Makefile        machine_pin.c    machine_wdt.c   modmachine.h  modutime.c      mpthreadport.h  tusb_port.c
README.md       machine_pwm.c    main.c          modrp2.c      mpconfigport.h  qstrdefsport.h  uart.c
fatfs_port.c    machine_spi.c    manifest.py     modrp2.h      mphalport.c     rp2_flash.c     uart.h
machine_adc.c   machine_timer.c  memmap_mp.ld    modules       mphalport.h     rp2_pio.c
max@MAXNB:~/esp/micropython/ports/rp2$ make
[ -d build ] || cmake -S . -B build -DPICO_BUILD_DOCS=0
CMake Error: The source directory "/home/max/esp/micropython/ports/rp2/build" does not exist.
Specify --help for usage, or press the help button on the CMake GUI.
Makefile:10: recipe for target 'all' failed
make: *** [all] Error 1
max@MAXNB:~/esp/micropython/ports/rp2$
```

how i do wrong ? Thanks !

robert-hh, 2021-03-26:
You did not create 'build' beforehand. You can also simply call `make`, which uses Makefile, set up as a wrapper around cmake.

maxpromer, 2021-03-26:
> You did not create 'build' beforehand. You can also simply call `make`, which uses Makefile, set up as a wrapper around cmake.

make `build` not work

```
max@MAXNB:~/esp/micropython/ports/rp2$ ls
CMakeLists.txt  machine_adc.c  machine_timer.c  memmap_mp.ld  modules         mphalport.h     rp2_pio.c
Makefile        machine_i2c.c  machine_uart.c   modmachine.c  moduos.c        mpthreadport.c  tusb_config.h
README.md       machine_pin.c  machine_wdt.c    modmachine.h  modutime.c      mpthreadport.h  tusb_port.c
build           machine_pwm.c  main.c           modrp2.c      mpconfigport.h  qstrdefsport.h  uart.c
fatfs_port.c    machine_spi.c  manifest.py      modrp2.h      mphalport.c     rp2_flash.c     uart.h
max@MAXNB:~/esp/micropython/ports/rp2$ make
[ -d build ] || cmake -S . -B build -DPICO_BUILD_DOCS=0
make -s -C build
make[1]: *** No targets specified and no makefile found.  Stop.
Makefile:10: recipe for target 'all' failed
make: *** [all] Error 2
max@MAXNB:~/esp/micropython/ports/rp2$
```

robert-hh, 2021-03-26:
Just enter
`make`
to run the build, or
`make clean`
to clean a previous build, whis is sometimes required.

maxpromer, 2021-03-26:
fixed by update cmake to latest.

closed 2021-03-26 by maxpromer

superbenji711, 2021-06-02:
running ```make [options] ..``` worked for me.