micropython/micropython #7079: ESP32: Unexpected high pulse on GPIO pin init

robert-hh, 2021-03-27:
This effect was mentioned by used @kotk_o on the forum. 
It is impossible to initialize a GPIO pin as low level output and keeping it's output low for the whole process. During init, there will be a high pulse of about 300µs. The reason is the call to gpio_reset_pin() during init, which sets the pin to input mode with the internal pull_up enabled.
Due to that, the GPIO settings do not allow a pin to be configured as input with high impedance. If no pull mode is specified, it stays in the reset mode, which is PULL_UP.
There was obviously a good reason to call gpio_reset_pin(), because that sets it to a known state. But the side effect is not always appropriate.

RCS-it1, 2021-03-29:
Please, read forum page: https://forum.micropython.org/viewtopic.php?f=18&t=10112

dpgeorge, 2021-04-15:
> There was obviously a good reason to call gpio_reset_pin(), because that sets it to a known state. But the side effect is not always appropriate.

See #5771 and commit 8a917ad2529ea3df5f47e2be5b4edf362d2d03f6 for this change.

dpgeorge, 2021-04-15:
See #7121 for a fix.

dpgeorge, 2021-04-16:
Should be fixed by e5d2ddde25351f1ede7d0d1b00be632301962fb0

closed 2021-04-16 by dpgeorge