micropython/micropython #7082: ESP32 GPIO wrong initial configuration. Pull-up always enabled

RCS-it1, 2021-03-29:
Hello Team,

We detected a **critical error** on micropython for ESP32 (I don't know if it appears in other ports). It seems that when you configure any GPIO, the PULLUP is always enabled during 150-200us. Doesn't matter if you configure the GPIO as input or output. 

This bug is confirmed from multiple users. See this link:

https://forum.micropython.org/viewtopic.php?f=18&t=10112

RCS-it1, 2021-03-29:
repeated topic..

closed 2021-03-29 by RCS-it1

robert-hh, 2021-03-29:
My PR was initiated by your forum topic. Since you did not raise one for a few days, I did that in your favor. Your initial problem was the high level, but having no option to set the GPIO as high impedance is also disturbing.

dpgeorge, 2021-04-15:
Duplicate of #7079.