micropython/micropython #7089: rp2: Device does not register on an USB2 port

robert-hh, 2021-03-31:
Coninuing the discussion of issue #6866
The RP2 dos not show up in bootsel mode or in normal mode with MicroPython when connected to a USB2 port.
MicroPython version MicroPython v1.14-128-gca3d51f12 on 2021-03-31; Raspberry Pi Pico with RP2040
If works with adafruit-circuitpython-raspberry_pi_pico-en_US-6.2.0-beta.4, and fails with
adafruit-circuitpython-raspberry_pi_pico-en_US-6.2.0-beta.1.
Using a USB3 port everything is fine. That happens on both Linux and Windows.
A few listings of lsusb below. Not that no external hub was conncted., The only other devices are the internal camera `ID 0c45:6492 Microdia` and the keyboard `04d9:1512 Holtek Semiconductor, Inc.`. The last lsusb cal shows the RP2 at `2e8a:0005`.
```
RP2 connected to a PORT labelled as USB2
Bus 003 Device 003: ID 0c45:6492 Microdia 
Bus 003 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 005: ID 8086:0189 Intel Corp. 
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 017: ID 04d9:1512 Holtek Semiconductor, Inc. 
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

RP2 connected top the same port, but Keyboard at a different plug
Bus 003 Device 003: ID 0c45:6492 Microdia 
Bus 003 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 005: ID 8086:0189 Intel Corp. 
Bus 001 Device 051: ID 04d9:1512 Holtek Semiconductor, Inc. 
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

ESP32 conncted to the same port as the RP2 before
Bus 003 Device 003: ID 0c45:6492 Microdia 
Bus 003 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 005: ID 8086:0189 Intel Corp. 
Bus 001 Device 056: ID 10c4:ea60 Cygnal Integrated Products, Inc. CP2102/CP2109 UART Bridge Controller [CP210x family]
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 017: ID 04d9:1512 Holtek Semiconductor, Inc. 
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

RP2 connected to a port labeled as USB3
Bus 003 Device 003: ID 0c45:6492 Microdia 
Bus 003 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 005: ID 8086:0189 Intel Corp. 
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 017: ID 04d9:1512 Holtek Semiconductor, Inc. 
Bus 002 Device 018: ID 2e8a:0005  
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

lurch, 2021-03-31:
Could you repeat the same tests but include the output of both `lsusb` and `lsusb -t` please?

And do you see anything new appear in `dmesg` when you plug the Pico into a USB2 port?

I've never come across "Rate Matching Hub" before, I wonder if that might have anything to do with it? https://linux-hardware.org/?id=usb:8087-0024 suggests it's fairly common though, so I wonder why we've not had anybody else (that I'm aware of) reporting problems with BOOTSEL mode. :man_shrugging:

lurch, 2021-03-31:
I've checked with a colleague, and he says it still sounds like the enumeration issue as mentioned at https://github.com/micropython/micropython/issues/6866#issuecomment-810829627
(it's likely there are 'internal USB devices' on the USB2 bus, that are triggering RP2040-E5)

So I guess you have two workarounds: always plug your Pico into a USB3 port, or always plug you Pico into a USB hub, and then connect the hub to a USB2 port.

robert-hh, 2021-03-31:
It's not a problem for me ti use an USB 3 port. And if the MicroPython firmware will eventually add a workaround like Circuitpython does, than the limitation only applies to the bootsel mode.

But I keep the PR open for people to find it, when they make the same observation.

P.S.: I had problems in using USB hubs for devices like PyBoard or ESP32. After a while the hub seemed to disconnect them.

dpgeorge, 2021-04-01:
Looking at CircuitPython, from what I can tell the main differences are that they enable the USB enumeration fix, and have updated to v0.9.0 of TinyUSB (or very close to that version).  Here we use v0.8.0.

@robert-hh can you please try #7090 to see if that fixes your USB2 issue?

robert-hh, 2021-04-01:
Yes, it does. This change makes the board visible on the USB2 port, once the firmware was loaded using the USB3 port. Obviously it does not change the bootsel mode behavior.

dpgeorge, 2021-04-06:
Should be fixed by 2d8aecd2ad43afeeb14fb8a21fcff29a1000c5c4

closed 2021-04-06 by dpgeorge