micropython/micropython #7095: Unable to compile for ESP32

br0kenpixel, 2021-04-04:
Hi. I'm trying to compile MicroPython for my ESP32 (w/ SPIRAM support).
Make fails with the following error:

```
CMake Error at /Users/xxx/Documents/pixelweather/micropython/new/micropython/ports/esp32/main/CMakeLists.txt:5 (include):
  include could not find requested file:

    //py/py.cmake
Call Stack (most recent call first):
  /Users/xxx/Documents/pixelweather/micropython/new/esp-idf/tools/cmake/scripts/expand_requirements.cmake:107 (include)
  /Users/xxx/Documents/pixelweather/micropython/new/esp-idf/tools/cmake/scripts/expand_requirements.cmake:217 (expand_component_requirements)

CMake Error at /Users/xxx/Documents/pixelweather/micropython/new/esp-idf/tools/cmake/idf_functions.cmake:338 (message):
  Failed to expand component requirements
Call Stack (most recent call first):
  /Users/xxx/Documents/pixelweather/micropython/new/esp-idf/tools/cmake/project.cmake:173 (idf_import_components)
  CMakeLists.txt:38 (project)
```
I'm not sure why it's trying to access **//py/py.cmake**. It's definitely not a valid path. It should rather be something like **../../py/py.cmake** or **/Users/xxx/Documents/pixelweather/micropython/new/micropython/py/py.cmake**.

xtensa compiler version: gcc_8_2_0 2020r1 macos
ESP IDF version: 3.x

dpgeorge, 2021-04-06:
> ESP IDF version: 3.x

The esp32 port new requires IDF v4.x.  I suggest using IDFv4.2.

br0kenpixel, 2021-04-07:
I updated to IDF v4.2.0 and it works now. Thank you!

closed 2021-04-07 by br0kenpixel