micropython/micropython #7096: [WIN32] arguments to .py-script

gvanem, 2021-04-04:
I have problems receiving a command line in my script. A `len(sys.argv) == 1` always for a short option.

A little test `micropython.exe test-script.py -h` does not seems to pass along `-h` to `test-script.py`.
But instead swallow it in `ports/unix/main.c` and `pre_process_options()`.

But `micropython.exe test-script.py --help` gets passed along. Working as designed?

stinos, 2021-04-05:
> A len(sys.argv) == 1 always for a short option.

Only for `-h` though? Others should work fine.

`pre_process_options` is too simple/naive here, in that it should only look at the arguments coming before the `file` argument. Like CPython it should treat `python [option] ... [-c cmd | -m mod | file] [arg]` such that `sys.argv[1:] == arg` no matter what.

gvanem, 2021-04-05:
> Only for -h though? Others should work fine.

Yes, that's correct. But I'd like my `test-script.py` to show some help too 🤒.

stinos, 2021-04-06:
> Only for -h though? Others should work fine.

Turns out: not really. E.g. `micropython test-script.py -X blabla` would exit immediately with `Invalid command line arguments.`

See #7100 for a fix

dpgeorge, 2021-04-07:
Thanks for the bug report.  Should be fixed by a66286f3a0f24985cae2648102e51de64b8a1871

closed 2021-04-07 by dpgeorge