micropython/micropython #7101: ESP32 CMake based build isn't compatible with berkeley-db

marcidy, 2021-04-07:
Trying to include btree with the new CMake based build system for ESP32 fails, and I suspect it's due to berkeley-db not being cmake compatible.

This is with idf v4.0.2 and micropython @ 4f53f462 (latest as of this post).  I don't think the idf version matters as it's more related to including berekley-db.

I enabled btree through boards/GENERIC/mpconfigboard.h:

```
#define MICROPY_HW_BOARD_NAME "ESP32 module"
#define MICROPY_HW_MCU_NAME "ESP32"
#define MICROPY_PY_BTREE (1)
```

The build fails looking for db.h:
```
/micropython/extmod/modbtree.c:37:10: fatal error: db.h: No such file or directory
 #include <db.h>
```

`db.h` is provided by `lib/berkeley-db-1.xx/PORT/include`.

I'm not sure the best approach to resolving this, especially given it's upstream. 

Is there a desire to fix this, or should be rely on building as a native module / user module?

dpgeorge, 2021-04-07:
> Is there a desire to fix this, or should be rely on building as a native module / user module?

Yes this should be fixed, it's really a regression moving from make to cmake because it was previously (with make) enabled for all esp32 boards.

It could be a user module... but that might introduce more complexity than needed at this stage.

marcidy, 2021-04-08:
I'm struggling with this, it's outside my knowledge of cmake and idf components.  While I'm focused on btree i think the solution will be applicable for all the extmods.

I can keep at it, but if anyone has pointers I'd welcome the help.

I see a number of issues:
1) lots of tweaking of the build for each external component
2) the external components aren't built as libs which might make this easier
    2.a) can't easily promote them to libs due to an invocation of cmake's scripting mode (possibly by idf or implicit with cmake include, not sure)
3) modbtree is written as if it's in the build directory, which tightly couples the code to the build structure
4) changing the tree or include structure of any module will break the make build

Here's some illustration based on btree.

`extmod.mk` does this:

```
ifeq ($(MICROPY_PY_BTREE),1)
BTREE_DIR = lib/berkeley-db-1.xx
BTREE_DEFS = -D__DBINTERFACE_PRIVATE=1 -Dmpool_error=printf -Dabort=abort_ "-Dvirt_fd_t=void*" $(BTREE_DEFS_EXTRA)
INC += -I$(TOP)/$(BTREE_DIR)/PORT/include
SRC_MOD += extmod/modbtree.c
SRC_MOD += $(addprefix $(BTREE_DIR)/,\
    btree/bt_close.c \
    btree/bt_conv.c \
    btree/bt_debug.c \
    btree/bt_delete.c \
    btree/bt_get.c \
    btree/bt_open.c \
    btree/bt_overflow.c \
    btree/bt_page.c \
    btree/bt_put.c \
    btree/bt_search.c \
    btree/bt_seq.c \
    btree/bt_split.c \
    btree/bt_utils.c \
    mpool/mpool.c \
    )
CFLAGS_MOD += -DMICROPY_PY_BTREE=1
# we need to suppress certain warnings to get berkeley-db to compile cleanly
# and we have separate BTREE_DEFS so the definitions don't interfere with other source code
$(BUILD)/$(BTREE_DIR)/%.o: CFLAGS += -Wno-old-style-definition -Wno-sign-compare -Wno-unused-parameter $(BTREE_DEFS)
$(BUILD)/extmod/modbtree.o: CFLAGS += $(BTREE_DEFS)
endif
```

and `modbtree` has this bit:
```
#include <db.h>
#include <../../btree/btree.h>
```

`../../btree/btree.h` makes sense given it copies all the src into the build directory, but i dont think these make sense in a more vanilla cmake build where berkeley-db would probably by a library.  idf components are basically libraries

There's customization of the build process needed for berkeley-db.  I can pick those up with target_compile_options, but this implies a library build.

With my limited understanding, I suspect the solution will be reworking the make build so it's more compatible with a cmake build (which impacts all ports, though hopefully it would be completely transparent).  But not sure, there could be tools in cmake that make this easier. 

I looked at cmake utils like external_projectadd and add_custom_command but it wasn't immediately obvious to me how to cross the barrier from external source to integrating with build like modbtree's includes expect.

So I think this is currently over my head (difficult for me, but not for someone else), or actually difficult.  I'm going to continue looking at it, but any help appreciated.

marcidy, 2021-04-08:
Ah, I see #7106 .  Thanks!

dpgeorge, 2021-04-09:
> So I think this is currently over my head (difficult for me, but not for someone else), or actually difficult. I'm going to continue looking at it, but any help appreciated.

I also found it difficult to do (but mainly because I don't really know cmake that well).  It definitely needed to be a cmake library component because the berkeley-db code needs special C definitions that, if applied to all source, mess up the build.  The main issue I had was that `extmod/modbtree.c` is the interface between berkeley-db and MicroPython/IDF and so needed to be built with C definitions from both sides of that interface.  In the end I resolved this by adding `MICROPY_DEF_CORE` with just the definitions needed to build `modbtree.c`.

dpgeorge, 2021-04-09:
Fixed by ab9d47e0230768f82a735343fa5b58e2029d8eda

closed 2021-04-09 by dpgeorge