micropython/micropython #7114: problem to uncompreess .tar file

tomasbaldi, 2021-04-12:
Hi! i have a problem when i try uncompress a .tar file with utarfile library. When execute this program, all files are descompress, but are empty.

import os
from lib.shutil import shutil
from lib.utarfile import utarfile

```
t = utarfile.TarFile('file.tar')

for i in t:
    print(i)
    if i.type == utarfile.DIRTYPE:
        os.makedirs(i.name)
    else:
        f = t.extractfile(i)
        shutil.copyfileobj(f, open(i.name, "wb"))
```

This is based on example-extract.py attached in utarfile library.

dpgeorge, 2021-04-13:
Can you provide an example .tar file that shows the problem?

jonnor, 2024-09-13:
Insufficient information provided to be able to reproduce/debug the problem.

closed 2024-09-13 by jonnor