micropython/micropython #7115: Behaviorof utime.ticks_diff() when using float as parameter

Pyr0m4ni4c, 2021-04-13:
Using the provided method ticks_diff(param1: int, param2: int) fromthe  utime-library an unexpected behavior can be observed when passing one or two floats as param1 or/and param2. 
[image]

## Expected Behavior
>>>utime.ticks_diff(2.0, 1)
>>>1

>>>utime.ticks_diff(2, 1.0)
>>>1

>>>utime.ticks_diff(2.0, 1.0)
>>>1

## Current Behavior
>>>utime.ticks_diff(2.0, 1)
>>>268475999

>>>utime.ticks_diff(2, 1.0)
>>>-268476014

>>>utime.ticks_diff(2.0, 1.0)
>>>-40

## Steps to Reproduce
1. import utime
2. call utime.ticks_diff(2.0, 1.0)
3. repeat step 2 to see different results

## Environment
MicroPython v1.13 on 2020-12-15; PDS4.0-Modules with STM32F405RG

dpgeorge, 2021-04-13:
This function does not support passing anything other than small-integers to it.  The only values that should be passed to `ticks_diff()` are values returned from `ticks_us()`, `ticks_ms()`, `ticks_cpu()` (this is explained in the docs: http://docs.micropython.org/en/latest/library/utime.html#utime.ticks_diff).

Pyr0m4ni4c, 2021-04-14:
Thank you very much for the quick response. 
Alright, should have read past the code example. (•_•)

It was just something I encountered when testing different timings specific to my use-case. Never expected this kind of behavior. So I'll throw the following question in the room and close the subject:
If it doesn't support float (or even big integer) values, shouldn't it then throw an exception? 

Again, 
thank you very much.

closed 2021-04-14 by Pyr0m4ni4c

dpgeorge, 2021-04-14:
> If it doesn't support float (or even big integer) values, shouldn't it then throw an exception?

The idea with these functions is that they should be efficient, and checking for invalid types is not efficient.  It doesn't actually do any harm to pass in non-small-ints (it will not crash), it just gives nonsense answers.

fragmuffin, 2023-05-30:
I found similar behaviour while using `int`.
I was experimenting with using an "error" value, that can still act as a somewhat valid timestamp (in theory), but the results were similar to using a float.

```python
>>> import utime as time
>>> TICKS_ERROR = 0xffffffff
>>> TICKS_MASK  = 0x3fffffff
>>> time.ticks_diff(TICKS_MASK, 0)
-1
>>> time.ticks_diff(TICKS_ERROR & TICKS_MASK, 0)
268495368
```

I learnt from this that `int`s aren't always the same size... which is obvious in C... but less so in python.
And because `ticks_diff` makes an assumption, even `int`s can cause this.

```python
>>> def test(a, b):
...     print(f"==:{a == b}; is:{a is b}: isinstance:{isinstance(b, type(a))}")
>>> test(1 + 1, 3 - 1)
==:True; is:True: isinstance:True
>>> test(1 + 1, 0xffffffff - 0xfffffffd)
==:True; is:False: isinstance:True
``` 

2 things to note from the last line:
- `0xffffffff` is a 32-bit int when unsigned, but micropython's internals will be storing this as a signed number... therefore will be using > 32-bits to store it... therefore:
- although the numbers are `==`, and are both of instance `int`, they do not share the same memory address (`id(a)!=id(b)`)

Interestingly: you can actively create 2 `int`s of the same value, forcing them to be stored in a 64-bit int instance.

```python
>>> test(array('q', [1])[0], array('q', [1])[0])
==:True; is:False: isinstance:True
```

reference: see [`struct` documentation](https://docs.python.org/3/library/struct.html#format-characters) for what `'l'` and `'q'` mean: 

### Workaround

This can be "fixed" by hard casting the number to use a 32-bit `int`

```python
>>> from array import array
>>> test(1 + 1, array('l', [0xffffffff - 0xfffffffd])[0])
==:True; is:True: isinstance:True

# or... in the context of this issue...

>>> cast = lambda x: array('l', [x])[0]
>>> time.ticks_diff(cast(TICKS_ERROR & TICKS_MASK), 0)
-1
```

But as @dpgeorge said, this will be at the cost of speed, and it's pretty messy.

I hope someone finds this useful.
Personally; for my project, I've decided this isn't a good design, or readable, or portable...
I'll just store a separate `bool` to store an error-state :wink: :+1:

robert-hh, 2023-05-30:
The first sentence of the documentation for ticks_diff() is:

> Measure ticks difference between values returned from [ticks_ms()](https://docs.micropython.org/en/latest/library/time.html?highlight=ticks_diff#time.ticks_ms), [ticks_us()](https://docs.micropython.org/en/latest/library/time.html?highlight=ticks_diff#time.ticks_us), or [ticks_cpu()](https://docs.micropython.org/en/latest/library/time.html?highlight=ticks_diff#time.ticks_cpu) functions, as a signed value which may wrap around.

From that it should be clear, that not arbitrary values can be supplied.